OttoDIY/blockly #4: Auto Updater

cparrapa, 2020-01-23:
Hi @BodoMinea 

It seems you are much better at code than me so i am just gonna put some challenges as issues, that i faced while developing this to solve, if you maybe know the solution or at least the way to do it, it will be greatly appreciated:

It seems the auto update is still linked to the forked project and is creating some confusion in users, i tried by changing name inside the packgage.json but i must be missing something because it was bad built. It must be connected to having links in other files.

Thanks!

BodoMinea, 2020-01-23:
Hi, maybe this file - https://github.com/OttoDIY/blockly/blob/ff51e88458ac6622e0796816e1e5eae22962d048/app-update.yml is the culprit?

I have no idea how to test the auto update mechanism but I will investigate.

cparrapa, 2020-01-23:
Maybe we can try by changing them, we can test with about button it has a ¨¨Check now¨¨ button and a check box to do it automatically every times restart

BodoMinea, 2020-01-24:
So, we should first fix the specified owner and repo in the referenced files (app update and dev app update).

Then, we should make a GitHub release for the current Otto Blockly version as instructed here - https://help.github.com/en/github/administering-a-repository/creating-releases, making sure the version number matches the current project version number in package.json (and this syncing should be always maintained when pushing releases, so that the client software knows when to trigger an update).

Once this is all figured out and tested, the software can ship with auto update on in order to make fixes easier to obtain for end users (eventually on a notification-based system).

By changing only the references in the aforementioned files and not pushing a release tag, this is the result:
[image]

cparrapa, 2020-01-25:
Looks like the right approach I got.to the point that a release was needed but this is the first time I make a software and in github, so all of this is pretty new for me. 

Maybe we can start with version 1.0?

cparrapa, 2020-02-12:
Hi @BodoMinea 

I just fix the auto update related files changing the repo links inside the .yml and package.json files at least we wont have the problem of updating from the wrong repo i guess.

You can check in this commit: https://github.com/OttoDIY/blockly/commit/ebca74879dbfb65f0d8bb5da0bf2d3ff488ac953

Now trying to generate the token to publish the app as a release(npm run publier). The access token should have the repo scope/permission. Define GH_TOKEN environment variable. (not sure how to do this last part). So far unsuccessful with this part.

I will just manually upload the windows installer 1.0.0. as a github release for this repo. to at least have a base for future update version.

Cheers

BodoMinea, 2020-02-16:
Sounds good, I'll test when I'm at my home computer.

cparrapa, 2020-02-16:
Yes i just did not manage to put the token in the right place i am trying to find where should i put it, and is not possible to upload manually the installer without having the whole thing as a release which is not optimal as just having the .exe files in the releases.

cparrapa, 2020-05-06:
it looks to me that now is solved in the last couple of releases

closed 2020-05-06 by cparrapa