OttoDIY/blockly #5: Serial Monitor

cparrapa, 2020-01-23:
Hi @BodoMinea

Other challenge is the serial monitor.

It seems now is not really working, i do not know what is missing too, actually i would like just something simple to test sensors input easy for kids. We can  even simplify and i can create the necessary remaining blocks

Thanks!

BodoMinea, 2020-01-23:
Hi!

I'll definitely look into the issue.
My first concern is the lack of a Serial Begin block.
[image]

I don't think you can write into serial without that.
Anyway, I manually added a Serial.begin(9600) in the code view and uploaded to the board and indeed: the Blockly serial monitor shows nothing, while I get the desired output in Arduino IDE or Putty.

Any ideas on how to properly debug the desktop app? I would need to see its console while properly running it as a packaged app in order to diagnose the built-in serial monitor. What's the platform we're dealing with here. ElectronJS?

cparrapa, 2020-01-23:
Ok  so i will work in a new serial begin block or integrate into that block, i saw it working in this other related blockly (https://github.com/technologiescollege/Blockly-at-rduino) so maybe we can just do the same.

BodoMinea, 2020-01-24:
Yes, the F8 trick worked for debugging but it's a little bit hard to track everything because it tries to console.log() in the system console, not browser console.

Anyway, I fixed it. There were two problems with the serial monitor:
- it was trying to read the baud rate from a non existing local storage variable (changing the speed did not change this value in local storage and also no fallback default was in place)
- it was ALWAYS trying to parse line by line, but sometimes serial is just a continuous string of data, so you cannot split and parse line by line - now it's using buffers and works for both Serial.print and Serial.println
[image]

Close this after you merge PR #8

closed 2020-01-24 by cparrapa