OttoDIY/blockly #7: Block category icons disappear after switching language from default

BodoMinea, 2020-01-23:
English:
[image]
French:
[image]
The icons should remain the same. Any idea why this is happening?

cparrapa, 2020-01-23:
It is just not updated, don´t worry but yes we have´re check all the lang files, some community members are maybe gonna help to add new languages, so then we can make them consistent. Or any suggestion how to deploy and easy collaborative way to translate?

BodoMinea, 2020-01-26:
Fixed, PR #10

closed 2020-01-26 by BodoMinea

cparrapa, 2020-01-26:
Thanks, 
@BodoMinea do you also experience that switching between languages, that sometimes does not work immediately? like is necessary to refresh (F5) or even restart the app and therefore deletes all current blocks? maybe we need something more instant or automatically for general users, just a thought. 

Similar situation with the setting (Language & Blocks menu) the level sometimes get mixed or showing the categories of other levels instead?

BodoMinea, 2020-01-26:
This is a separate issue.