OttoDIY/blockly #9: Mac and Linux installers

cparrapa, 2020-01-25:
So I have been trying really hard to build from my Windows computer using electron, the Mac and linux versions, unfortunately the application that is generated doesn't open or is just white. 

My dream will be to make it also for raspberry pi so that we could make everything with opensource hardware! 

There must be a way since electron.is is supposed to be a multi platform development environment for almost all OS.

Alternatively we could "just " find a way to make the arduino uploader work for this online app, for chrome books used in schools this will be amazing.  Before in other blockly arduino repos, online uploading was working with codebender.cc but it seems is no longer available or more of a private project not sure. Maybe we could upload via Bluetooth using .hex files will be even easier than USB drivers ?

I don't know, just brainstorming @BodoMinea 😄 putting in common the major challenge to make this software really available to anyone. 

You have solve the other issues, so fast, impressive👏so happy to have your support, thanks a lot from all the community, I am pretty sure they are happy too.

Have a good weekend.

BodoMinea, 2020-01-25:
We cannot simply make the web version upload to the Arduinos because it has no access to hardware serial ports. There are however some approaches to make webapps able to compile and upload the code. For example, offer a small „adapter” software, like, a NodeJS script or even Chrome extension which accesses the hardware ports and communicates by local web requests with the online Blockly.

However, for all intents and purposes I think we should build proper installers for Mac (pkg->app or directly app) and Linux (a sh installer script or distribution packages). I don't think it's that hard and most probably I am able to do that. It will take some more time since I will be away with work the next week.

I have several Raspberry Pi's and a friend's Mac at my disposal so I can maybe try to build from source and package directly on the target hardware.

cparrapa, 2020-01-25:
Cool! I also have one raspberry pi, yes take your time, let me know if you need something. I will be testing more the new PRs, new blocks that I have been working on and see how to add Spanish language, with that we will learn how to add more.

BodoMinea, 2020-01-25:
Everything you need to change/add for multi-language resides in www/lang.

In code.js you define new languages or new spots in the app for translatable string, while for the translations themselves you need to add more of or edit the existing COMPONENT_LANG.js file, like Arduino_en.js.

I am also planning to translate this to Romanian, which is my primary language, when I´ll have some more spare time to go through all the strings.

cparrapa, 2020-01-26:
Thanks for the tip, this is what a member of the community did so far https://gist.github.com/0day1day/20130c46310f396051e88c42921164d8

Then I guess I have to define the language in code.is as you said.
Good to have Romanian too.

But this is very manual I am just concern if we need a better translation system once we add more blocks and make changes in the main one how to keep it consistent and updated.

BodoMinea, 2020-01-26:
I added the Spanish translation based on @0day1day´s work. A few interventions were necessary:
- currently, the translation module will crash if there are not all the required files present. So given that only the Arduino translation is given in the gist you forwarded, I duplicated the Blockly and Message files that are in English over for Spanish - so now we got a partial translation working.
- the file had some minor formatting errors (related to colons, command, etc). When translating so many strings that is expected to happen. If a more intuitive translation system was in place, as you suggested, this would be mitigated. However, due to the diverse nature of translation systems, I do not know of any ready-made product, open source or otherwise that would provide an editing frontend for the internalization format present in blockly. So we will have to come up with our own soution for the future unless someone has a better idea.
- I added the references in code.js so that this will show up in the selection menu.

Test it live: https://bodominea.github.io/blockly/www/ or see PR #10

s1gmund80, 2020-02-12:
Hi, 
is there something I could help with to have a proper Mac build installer/app? 
I have a Mac, I can do some tests and maybe help with some build/install script

cparrapa, 2020-02-12:
Hi @s1gmund80 
Yes of course, So i have been trying to follow this tutorial https://www.christianengvall.se/dmg-installer-electron-app/
and i generated one for MAC already here:
https://www.ottodiy.com/#blockly
Download try and let me know.

If not that means i miss something, which we should find a way from the electron documentation https://www.electronjs.org/docs/tutorial/application-distribution
or here https://www.electron.build/multi-platform-build

Thanks!

nickboucart, 2020-02-13:
I too have a Mac. I tried the installer, but that didn't work, the app wouldn't startup up.

I cloned the project and try to run it locally, but all I get is a blank screen... Do you have any suggestions on how to get the app running locally in development? If I could get that to work on my Mac, I might be able to look into installers etc...

Thanks!

cparrapa, 2020-02-13:
Hi @nickboucart 

That is a great progress! at least we know the app runs well just need to figure out how to pack in an APP.

The blank screen was happening to me also, in my case it was because i did not open git bash as admin.

For MAC i think there must be another process also to build.

Maybe we need to modify the packgage.json file so that it builds all the OS together with one command like the shortcuts in this tutorial https://www.christianengvall.se/electron-packager-tutorial/

Thanks!

nickboucart, 2020-02-13:
Hi,

The app is not really running well on macosx as it stands now: I browsed
through the source code and found couple of places where windows specific
calls where made (e.g. calls to the arduino.exe and python.exe). These will
have to be made os dependant.

I’m completely new to electron development, so I have to find my way around
a bit.I might have some time next week to look into the mac side of things.
I'm not promising anything though ;)

Nick.

cparrapa, 2020-02-16:
Hi @nickboucart and @s1gmund80

I am also new in this electron, java, Blockly development world and still with help of people and just trying we manage to make a very functional windows app. I only have Windows computers so i cant verify whatever we output for other OS is fully working. So you guys a big help!

You are right about the calls maybe we need the arduino cli compiler version for linux and mac as well, it is not that simple as  just pack the app for other OS with electron.

Thanks intentions count!

nickboucart, 2020-02-17:
Hi all,

I'm working on getting the app running and working on mac. You can follow up on progress at https://github.com/nickboucart/blockly/tree/mac. That said, I don't know how much time I can spend the next couple of days/weeks.

Things I did so far
- get the app started in development mode on mac (had to change a number of paths in files for that)
- looked at compiling a sketch on mac. For that, I looked into arduino-cli for both installing all the boards and the toolchain, and for doing the actual compilation. I have this more or less figured out on my machine, but nothing near production ready.

I'll keep using this thread to post updates as I go.

Thanks!
Nick.

mxochicale, 2020-06-27:
Hi @cparrapa et al.,

Here to offer my weekends to help on the linux installer (Ubuntu, raspbian, etc) for Otto Blocky. From your above comments, I read that @BodoMinea suggested "(a sh installer script or distribution packages)" and I see that @nickboucart has done a nice progress with the [mac](https://github.com/nickboucart/blockly/tree/mac) installer.  That said, I am wondering if you have any references, links or instructions to understand more about the workflow on the linux installer for Otto Blockly. 

PS. You might create a new issue to follow up the linux installer and rename the current issue to mac installer only.

cparrapa, 2020-06-28:
Hi @mxochicale 

Indeed we better separate this issue for Linux here #38

Thanks for your help we are gonna need a lot i am also getting more time now for Blockly so in this July we can make some progress!, 

I did not see @nickboucart, advances is it something we can merge?  to be honest i am lost of what to do to make it work for Mac, i tried all in my humble software developing experience, this is what i used last time to pack it.

$ electron-packager . --overwrite --platform=darwin --arch=x64 --icon=build/icon.icns --prune=true --out=release-builds
Packaging app for platform darwin x64 using electron v4.1.1
Wrote new app to release-builds\Blocklino-darwin-x64

Another thing is that i do not have a Mac so very difficult for me to test, i gave to people to try but they only got white screens.

NeftaliVenancio, 2020-10-17:
Tengo equipo Mac e interesado en poder realizar pruebas.

cparrapa, 2020-10-20:
Hi @nickboucart Neftali
have you tried to generate the installer from electron builder?

cparrapa, 2021-10-30:
Hi guys use this one https://drive.google.com/file/d/1owVPRWOEcMCEG8Oo4l11p0zWlsCsBPOu/view

closed 2021-10-30 by cparrapa