OttoDIY/blockly #19: Mac, Linux and Windows installers in the Releases section of github

vongomben, 2020-02-16:
At the moment if you need to download the latest installer version you have to click / go through these  three links

--> https://www.ottodiy.com/#blockly
--> https://uploads.strikinglycdn.com/files/6c9116e7-80ff-4288-917b-831c4b8d01d5/Otto%20Blockly%20download.pdf
--> link from drive

My porposal is to host the releases on github [(in the releases section](https://github.com/OttoDIY/blockly/releases)), in this way people would easily see the version they are installing and the link would be easily tracked and shared (other than Google Drive)

cparrapa, 2020-02-16:
Hi @vongomben 
Yes but we are trying to find the way to fix the auto updater for this on issue #4 , without this we are not be able to pack the installers in the release section. 

The release section as is now, only allows us to release online version (as you see the present repo), so it won't work as an offline software release. 

Mac and Linux read #9

cparrapa, 2020-03-22:
I managed! now you can check here https://github.com/OttoDIY/blockly/releases/tag/v1.1.0
@vongomben, hope things are going good in your country.

closed 2020-03-22 by cparrapa