OttoDIY/blockly #22: Mouth - display variable

brendonhatcher, 2020-02-28:
I created an int variable and declared it. I added the Mouth block and tried to connect the variable block to it. Instead of showing the number, the led matrix displays the name of the variable.

The variable is storing an integer value. If I use the Pixel X/Y block, I can use the value of the variable to input the X and Y coordinates.

The "S" in the attached image is Otto scrolling the word CLAPS

[image]

cparrapa, 2020-02-29:
Oh i got your idea is pretty cool you want Otto to turn led by led every time you clap.

cparrapa, 2020-03-21:
ok solved here https://github.com/OttoDIY/blockly/commit/eb144a958a705dfbbeeddb8ab79ae72d074bc9a9

closed 2020-03-21 by cparrapa