OttoDIY/blockly #23: Simplfy Check and Upload

brendonhatcher, 2020-02-28:
The current process of pressing Check, then closing the popup, then pressing Upload, then closing the popup is very annoying when you are making lots of edits.  

It would be great if pressing it would do both the check code and upload and then close the popup. 
The check code button should remain for users to check without uploading.

This change would match the Arduino IDE user interface.

cparrapa, 2020-03-12:
You are right still trying to figure out how to solve this, it seems blockly needs to separately do both task.

cparrapa, 2020-03-22:
it was not easy to fix! but somehow i think i managed here https://github.com/OttoDIY/blockly/commit/59b7edae2e709e1103305912daaf00739ff8b3e2

closed 2020-03-22 by cparrapa