OttoDIY/blockly #24: Bluetooth connection

brendonhatcher, 2020-02-28:
Is it possible to connect OttoBlockly to my Otto Humanoid via bluetooth instead of a USB cable?

brendonhatcher, 2020-02-28:
@OttoDIY responded on facebook: good idea for a new feature, we tried to make it at least using the software serial communication it should be possible (like a replacement of the APP control), connect is easy, but programming is more complex that it sounds.

cparrapa, 2020-03-21:
@brendonhatcher  it is very complicated actually, i myself tried even just with Arduino IDE and still not success https://www.instructables.com/id/Arduino-Bluetooth-Programming/

cparrapa, 2020-09-28:
Hi for this we have create Otto Scratch AI https://www.ottodiy.com/academia

closed 2020-09-28 by cparrapa