OttoDIY/blockly #28: Mac version doesn't work on Catalina

BjornB2, 2020-03-31:
I've downloaded the mac version. If I unzip it with the default unarchiver in MacOS it doesn't allow me to run the app. If I unzip with "The Unarchiver" I can start the program, get the expected security stuff. After I allow the app to open the app crashes immediately.
Tried on two different macs. 

```
Error report:Process:               
﻿
Blocklino [85542]
Path:                  /Applications/Blocklino.app/Contents/MacOS/Blocklino
Identifier:            com.electron.blocklino
Version:               1.5.0 (1.5.0)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Blocklino [85542]
User ID:               501
﻿
Date/Time:             2020-03-13 10:14:52.482 +0100
OS Version:            Mac OS X 10.15.3 (19D76)
Report Version:        12
Bridge OS Version:     3.0 (14Y908)
Anonymous UUID:        0A837EB5-2BDF-8C21-C1C5-4CF871A04AC8
﻿
Sleep/Wake UUID:       7768A977-DB68-4529-B50B-B23D03819DA1
﻿
Time Awake Since Boot: 230000 seconds
Time Since Wake:       4300 seconds
﻿
System Integrity Protection: disabled
﻿
Crashed Thread:        0
﻿
Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY
﻿
Termination Reason:    DYLD, [0x1] Library missing
﻿
Application Specific Information:
dyld: launch, loading dependent libraries
﻿
Dyld Error Message:
  Library not loaded: @rpath/Electron Framework.framework/Electron Framework
  Referenced from: /Applications/Blocklino.app/Contents/MacOS/Blocklino
  Reason: no suitable image found.  Did find:
	/Applications/Blocklino.app/Contents/MacOS/../Frameworks/Electron Framework.framework/Electron Framework: file too short
	/Applications/Blocklino.app/Contents/MacOS/../Frameworks/Electron Framework.framework/Electron Framework: stat() failed with errno=1
﻿
Binary Images:
       0x1077c4000 -        0x1077c4ffb +com.electron.blocklino (1.5.0 - 1.5.0)  /Applications/Blocklino.app/Contents/MacOS/Blocklino
       0x10fcf3000 -        0x10fd83cb7  dyld (733.8)  /usr/lib/dyld
    0x7fff3a0c2000 -     0x7fff3a0c2fff  com.apple.Cocoa (6.11 - 23)  /System/Library/Frameworks/Cocoa.framework/Versions/A/Cocoa
    0x7fff3d6fe000 -     0x7fff3dac6ffc  com.apple.Foundation (6.9 - 1674.114)  /System/Library/Frameworks/Foundation.framework/Versions/C/Foundation
    0x7fff3de3e000 -     0x7fff3dee1ffb  com.apple.framework.IOKit (2.0.2 - 1726.80.1)  /System/Library/Frameworks/IOKit.framework/Versions/A/IOKit
    0x7fff47690000 -     0x7fff479e2ffa  com.apple.security (7.0 - 59306.80.4)  /System/Library/Frameworks/Security.framework/Versions/A/Security
    0x7fff48832000 -     0x7fff4889cfff  com.apple.SystemConfiguration (1.19 - 1.19)  /System/Library/Frameworks/SystemConfiguration.framework/Versions/A/SystemConfiguration
﻿
Model: MacBookPro13,2, BootROM 264.0.0.0.0, 2 processors, Dual-Core Intel Core i5, 2,9 GHz, 8 GB, SMC 2.37f21
Graphics: kHW_IntelIrisGraphics550Item, Intel Iris Graphics 550, spdisplays_builtin
Memory Module: BANK 0/DIMM0, 4 GB, LPDDR3, 2133 MHz, 0x80CE, 0x4B3445364533303445422D45474347202020
Memory Module: BANK 1/DIMM0, 4 GB, LPDDR3, 2133 MHz, 0x80CE, 0x4B3445364533303445422D45474347202020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x157), Broadcom BCM43xx 1.0 (7.77.106.3 AirPortDriverBrcmNIC-1440.1)
Bluetooth: Version 7.0.3f5, 3 services, 27 devices, 1 incoming serial ports
Network Service: RNDIS/Ethernet Gadget, Ethernet, en9
Network Service: Wi-Fi, AirPort, en0
USB Device: USB 3.0 Bus
USB Device: Apple T1 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 41.1
Thunderbolt Bus: MacBook Pro, Apple Inc., 41.1
```

cparrapa, 2020-03-31:
Thanks for the report @BjornB2 we have no idea how to solve this, we have been researching and is still a challenge to make the apple stable and functional. We need more Electron/Apple developers to help the project

cparrapa, 2021-10-30:
Use this one https://github.com/takujikawata-pr/blockly/tree/feature-mac-dev

closed 2021-10-30 by cparrapa