OttoDIY/blockly #31: alternative solution for mac and other platforms (Web USB)

cparrapa, 2020-04-03:
@brendonhatcher suggestion to Mac and Linux users:

Use Arduino Nano 33 IOT Module (same pins, still fits the project board) in order to use [WebUSB](https://github.com/webusb/arduino) with a [web version of Blockly](https://groups.google.com/forum/#!topic/blockly/XHDJhbjoNPE) making it [compatible with all platforms.](https://developers.google.com/web/fundamentals/native-hardware/build-for-webusb)

cparrapa, 2020-06-30:
Max in https://wikifactory.com/+OttoDIY/forum/category/NzI?thread=MzYxNTc1 says

¨¨i find that Nano mega328 is not on the supported list of webUSB devices - with the implication something in the architecture will not allow/ support webUSB. https://github.com/webusb/arduino/﻿
So the already suggested solution here of using a supported board is likely to be the simplest solution. Although Nano 33 IoT is not as cheap a board.
I will dig into the Architecture requirements of webUSB and also what architecture the Nano mego328 has in this regard.
﻿
An important factor here is that the www.arduino.create.cc web IDE will connect with a nano mega328 no problem on both Chromebook and Macos - so it can be done, and Arduino have the code !¨¨

Meanwhile i am intrigued as to why webUSB cannot connect to nano mega328 yet the Arduino Web Editor can.
On Chrome OS the Arduino Web Editor is made up of a Chrome OS Extension for some local config - I figure that reverse engineering this extension will show some interesting info.  It loads a background page ( similar tactic to webUSB ) - however here it has a wider array of what it terms 'board Flashers' including descriptions for AVR boards and the corresponding Functions.  

It could be that the whole serial port functionality is set out in plain old js in the Extension. I will take a more forensic look and post up some samples of the code i find.  Someone else may spot something they recognize in serial port communications.

So initial conclusion to the OP : USB comms for Blockly
The solution for browser based Blockly would be to follow the lead of Arduino in the Web Editor. That would mean developing a Chrome OS Extension for Chromebooks ( Arduino's version is 26k lines of code ) AND for MacOS a similar Plug In users would have to install. 

That is what Arduino have done and it works.  Their Create Plug In code is available on GitHub...  Their Chrome OS extension is in plain JS. Both would require some considerable work to port to support Blockly.
I will look through the Chrome extension code to see what can be cut out and how generic it is, at least that will give a view of how much work would be involved to port it.

This one somehow it works with Arduino nano https://lab.open-roberta.org/﻿
They have a pre-installed driver on the Nano board: https://jira.iais.fraunhofer.de/wiki/display/ORInfo/Set+Up+Nepo4Arduino﻿
They also do the same as Arduino, which is to install a Plug-In on the host OS: https://jira.iais.fraunhofer.de/wiki/display/ORInfo/Open+Roberta+Connector
﻿

cparrapa, 2021-06-06:
This has been solved in the new Online version that uses Arduino Create Plugin 
Give a try here!
https://ottoschool.com/blockly

closed 2021-06-06 by cparrapa

Haxim, 2022-01-17:
I might be missing something obvious, but what "agent" do I have to have installed on a chromebook in order to have https://ottoschool.com/blockly be able to see a Nano to upload code?

cparrapa, 2022-01-21:
Hi @Haxim 

For the online version you need to:

1. Download and install Arduino Agent
2. Find the config.ing file in the folder where Arduino Agent was installed
On macOS ~/Applications/ArduinoCreateAgent/ArduinoCreateAgent.app/Contents/MacOS/config.ini
On Linux ~/ArduinoCreateAgent/config.ini
On Windows C:\Users\[your user]\AppData\Roaming\ArduinoCreateAgentS
3. Open and modify the origins to "origins = https://ottoschool.com";
4. Save and make sure the name and extension of the file has not been changed "config.ini" config.ini.txt wont work for example
5. Go to https://ottoschool.com/blockly
6. Open the walk Example.
7. Select Arduino Nano old bootloader
8. Connect your Otto and Click in the USB icon
9. Check that Agent Installed: Yes and Agent Connected: Yes
10. Click the upload icon

Wait a bit and your Otto will be walking!.

The first 5 steps only need to be done one time (setup).

Kabirousseini, 2022-12-08:
How to modify the config.ini file on Chrome OS ?