OttoDIY/blockly #35: Updating RGB NeoPixel block in LED section with current Adafruit library name

wendtbw, 2020-05-26:
When I create the block *RGB NeoPixel" in the LED block section of blockly, The code blockly creates is using library name <NeoPixel.h> which I think is maybe old?. The new library from github is Adafruit_NeoPixel.h> 

PS.  I am a bit new but I really want to learn where in the blocky code, how the blocks are created in Blockly and how that block is programmed to create the Arduino code.  If its possible just want to know how that process is done in the Blockly code.  Thanks so much!   My dream is to be able to understand the code better and be able to submit my own pull requests to the otto blockly open source project!

cparrapa, 2020-06-28:
Hi 

Thanks this we can start working on next week so to resume:
1. New block in the LED section that is for a single NeoPixel, in your case 8mm but it can be for any size.
2. Add Adafruit library
3. Setting up the NeoPixel library, strips change the third parameter --
Adafruit_NeoPixel pixels(NUMPIXELS, PIN, NEO_RGB + NEO_KHZ800);

Anything missing?

The process for blocks is explained here https://developers.google.com/blockly many people ask for this but then they see is not so easy so they give up :D

So do not worry if you give me the exact parameters, output code an look of the blocks i will do it for you.

wendtbw, 2020-06-30:
Hi,  thank you for your support and help!

Yes, your list is correct.

This might be a bit too much and maybe confusing for some but Adafruit has a recommendation in the case of using single push through NeoPixel LED.  Just a side note for wiring

> With through-hole NeoPixels (5mm or 8mm), add a 0.1 µF capacitor between the + and – pins of EACH PIXEL. Individual pixels may misbehave without this “decoupling cap.”

It is a wiring note to add a capacitor.

Thank you for the link to the developers code!  I forked the OTTO Blockly repo and I will try to do some testing on the forked repo!  It might be hard but I will try!!

Thank you again @cparrapa

wendtbw, 2020-07-01:
@cparrapa 

I was just doing some testing today and learned something new.

Sorry I think I did not fully understand the third parameter setting.  RGB or GRB paramater

Adafruit recommends using the RGB parameter for the setting.....BUT in the test program I tested GRB and it works fine...but the color parameters in the code are executed and read different.  RGB parameter as (255, 0, 0) sets the color to RED.  GRB paramater as (255, 0, 0) sets the color to GREEN.  Just reads the color codes differently.  

On the Adafruit site it is saying to use RGB in this case so I would recommend using that setting in the third paramater.

Sorry for the confusion.

==============================================================================

Just for an example of the code.   This is really simple code to make the 8mm single Neopixel green.

``#include <Adafruit_NeoPixel.h>
#define PIN 1
#define NUMPIXELS 1

Adafruit_NeoPixel pixels = Adafruit_NeoPixel(NUMPIXELS, PIN, NEO_GRB + NEO_KHZ800);

void setup() {
pixels.begin();
pixels.show();
}

void loop()
{
pixels.setPixelColor(0,255,0,0);
pixels.show();
}``

I have tested this and it works great.

At first just for my learning I will make a new block in my forked repo using this code.   

Thank you again @cparrapa

cparrapa, 2020-07-01:
Hi @wendtbw 

Thanks for the support and the extra hand.
To make this simple I have design this simple table to help us create the blocks accordingly just add a new row with the new block or blocks and if they are were aorganized  i can easily add that into Blockly https://docs.google.com/spreadsheets/d/1E_wf6ls3sdqTvQDtcUtNwovEI5ymoyx0vgsi4uC-0f8/edit?usp=sharing

The key is to have mini codes that you test in Arduino and works and those functions can be parameterized into the blocks

After you get to know the workflow then it will be easier for you to contribute directly.

wendtbw, 2020-07-02:
Thank you @cparrapa .   

I understand!  that is a great system!

cparrapa, 2020-07-25:
We did it! solved in 664059259fbbe96d3b1395f61394c1f1d08c83c0

closed 2020-07-25 by cparrapa