OttoDIY/blockly #36: HELP need to add the new Arduino Nano Every board

cparrapa, 2020-06-03:
is an actual continuation of the issue #2 but specifically for the new Arduino Nano Every https://store.arduino.cc/nano-every-pack

i get to the point that i add the definitions in the boards.txt but i still get error when compiling
[image]

Any clues @BodoMinea ?

BodoMinea, 2020-06-03:
I think you should try installing the Every boards in a separate Arduino install and then copy what´s missing to the Otto Blockly compiler resources. I am pretty sure I can do it but I currently don´t have the time to contribute.

cparrapa, 2020-06-03:
Hi @BodoMinea 
Yes good idea i was trying that using the Arduino IDE files that have the Every functional and copying them to the compiler, but it just got worse damaging the opiboot and other nano boards.
[image]
[image]
Later on I even managed to compile one time but then i try to upload and 
[image]

I think we are close but i am not sure what toolset i need to modify or add there.

Thanks for your time like this giving me advice and tips it will greatly help us.

cparrapa, 2021-01-04:
Fixed in #76  thanks to @logix5

closed 2021-01-04 by cparrapa