OttoDIY/blockly #37: Driver install failure of CH341SER.INF

mxochicale, 2020-06-27:
Hi all,

As suggested in the [readme](https://github.com/OttoDIY/blockly#installer), I downloaded the [latest version](https://github.com/OttoDIY/blockly/releases) of otto blocky and installed OttoBlockly-Setup-1.2.0.exe and also tried OttoBlockly-1.2.0.exe in a windowsX64 machine. However none of them recognise the USB device. So I am trying to execute SETUP.exe downloaded from the suggested [path: pilotes/_CH341](https://github.com/OttoDIY/blockly/tree/master/pilotes/_CH341) which results in the following installation failure: 

[image]

I might be missing something. I am wondering if you any recommendations?

cparrapa, 2020-06-28:
Hi @mxochicale 
Good to see you here! driver hmmmm did you try to install from here  https://sparks.gogo.co.nz/ch340.html ? some computers are picky with the CH340 devices like the nano. 

What Windows are you running? even in my Windows 10 i did not need to install and the USB was recognized automatically.

mxochicale, 2020-07-03:
Many thanks Camilo, finally here to mention that I was not connecting the USB cable to OTTO which might be the reason to have the driver installation error. So after connecting it, the driver is successfully installed under window 10 Home version 1803 and I can happily play with some of the nice examples you show [here](https://www.youtube.com/watch?v=chcWxh4Co_c) :) 

[image]

I close the issue now but I guess feel free to re-open it if you or anyone has similar problems.

closed 2020-07-03 by mxochicale