OttoDIY/blockly #38: Linux and possible Raspberry Pi installer

cparrapa, 2020-06-28:
Separated from #9 as suggested by @mxochicale

cparrapa, 2020-08-16:
Hola @mxochicale

Nuestro amigo @agomezgar ya hizo un gran progreso aquí  https://github.com/agomezgar/masayloBlockly/tree/05e43fb3a85b9adc2c1ab78e9b46a361a9bdb510/linux 

Lo puedes ensayar?

mxochicale, 2020-08-16:
Wow, Gracias @cparrapa  y @agomezgar :)  Lo checo el pr'oximo fin de semana :)

gitaroktato, 2020-09-07:
Hi @mxochicale, is someone looking at this? Is there any progress? Are there any manual steps available for experimenting?

mxochicale, 2020-09-07:
Hi @gitaroktato 

> , is someone looking at this?

I am interested in this issue but making little progress on my weekends. 

>  Is there any progress? Are there any manual steps available for experimenting?

Antonio Gómez has made some progress [here](https://github.com/agomezgar/masayloBlocklyLinux) and waiting for some instructions stated [here](https://github.com/agomezgar/masayloBlocklyLinux/issues/1). Perhaps, you don't have to wait and have it a go.

gitaroktato, 2020-09-07:
@mxochicale after bumping serialport dependency version to 8.0.4 the build was successful, but when running the application with `npm install` I get a blank screen. Any guidance on how I can fix this?

P.S.: If this works, I can submit a PR.

gitaroktato, 2020-09-07:
It's not ready yet, but it's something. :+1: 

[image]

mxochicale, 2020-09-07:
Great [progress](https://github.com/gitaroktato/blockly/commit/0e56fe0be2fba0c0d69a8289d6a27ccc5936902a) @gitaroktato 

It would be nice that you add explicit instructions on how you are building it in Ubuntu plus the version and architecture of your GNU/Linux distro. Perhaps, this might be one [example](https://github.com/fontainejp/blocklino/#how-to-use) for the readme file. I would be more than happy to help to review it and test it on my machines.

agomezgar, 2020-09-07:
I'm deeply sorry. You're right. I must make some documentation tasks.
Problem is, i'm a high school teacher in a country deeply affected by covid
these data, and i'm too busy just now(please forgive my poor english). Your
first problem: if you upgrade your electron versión, you must add
webPreferences:{nodeIntegration:true} in every new BrowserWindow command
used in the electronApp.js file

Written from my smartphone (not so smart).

gitaroktato, 2020-09-08:
I'll just keep updating the issue, so someone can also take this further if I'm unable to.
You can track the progress by comparing master to my personal branch:
https://github.com/OttoDIY/blockly/compare/master...gitaroktato:linux_build

I use Ubuntu 18.0.4.5 LTS for building the project.

**Steps taken so far**
* Make sure you're having Python 2.7 installed (I use [miniconda](https://docs.conda.io/en/latest/miniconda.html) to create a separate environment for building)
* Make sure you have Node version > 10.0
* Change `serialport` version to 8.0.4 in `package.json`
* Change build arguments for `compiler` and `publier` commands to `--linux --x64`, but I haven't used those.
* Fix path locations in `electronApp.js` according to the changes above

**Building the project**
First, build & compile everything
```shell
npm install
```

Run electron app
```shell
npm start
```

gitaroktato, 2020-09-10:
@agomezgar @mxochicale  - could you please check out [this branch](https://github.com/OttoDIY/blockly/compare/master...gitaroktato:linux_build) and see if it works with the steps above?
All you need to do to run the application is an `npm install` followed by an `npm start`.

I can work on fixing the `compiler` and `deb64` goals in the meantime.

mxochicale, 2020-09-13:
Great progress @gitaroktato 

I have tried your linux_build branch in https://github.com/gitaroktato/blockly and I got the following error that might be related to some permissions. Any idea? 

[image]

Additionally, perhaps you would like to create a README file in ~/compilation/dependencies/ where you can start adding the above instructions. I think such instructions are very important to allow other users to test your work. For example, you can add in that README file something like this:
```
## usage 
git clone https://github.com/gitaroktato/blockly
cd ~/blockly
git checkout linux_build 

## dependencies and requirements 
* You can use my README on how to setup a conda env
https://github.com/mxochicale/BlocklyDuino/tree/49afe53fbc62f06a5c342e334a13c6b46581246a/setup_arduino_web_server

* Be more explicit in the installation of 
Node 10.22.0 # here some nice instructions https://www.digitalocean.com/community/tutorials/how-to-install-node-js-on-ubuntu-18-04
and 
npm 6.14.6 # here some nice instructions  https://github.com/nvm-sh/nvm

Also the first time I run npm install I need to run `sudo chown -R $USER:$(id -gn $USER) /home/$USER/.config`
```

mxochicale, 2020-09-13:
Thanks @agomezgar  for such great effort and share your work 

Slightly similar as above, and with the dependencies suggested by gitaroktato, I have tried your linux build in my machine with Ubuntu 18.04.3 LTS x64 - Linux kernel 5.3.0-53-generic by cloning https://github.com/agomezgar/blockly and I got the following error that might be related to some permissions. Any ideas on how to sort it out?

[image]

You might also like to create a **README** file as suggested [above](https://github.com/OttoDIY/blockly/issues/38#issuecomment-691623904).

agomezgar, 2020-09-13:
I could solve (or so I think) the serial monitor  problem, @gitaroktato . New versions of serialmonitor don't admit functions anymore. I.e., this old code:
`		sp.list(function(err,ports) {
			var nb_com = localStorage.getItem("nb_com"), menu_opt = portserie.getElementsByTagName('option')
			if(ports.length > nb_com){
				ports.forEach(function(port){
					if (port.vendorId){
						var opt = document.createElement('option')
						opt.value = port.comName
						opt.text = port.comName
						portserie.appendChild(opt)
						localStorage.setItem("com",port.comName)
					}
				})
			localStorage.setItem("nb_com",ports.length)
				localStorage.setItem("com",portserie.options[1].value)
			}
			if(ports.length < nb_com){
				while(menu_opt[1]) {
					portserie.removeChild(menu_opt[1])
				}
				localStorage.setItem("com","com")
				localStorage.setItem("nb_com",ports.length)
			}`
Must be changed for something like:
`sp.list().then(ports =>  {
			var nb_com = localStorage.getItem("nb_com"), menu_opt = portserie.getElementsByTagName('option')
			if(ports.length > nb_com){
				ports.forEach(function(port){
					if (port.vendorId){
						var opt = document.createElement('option')
						opt.value = port.path
						opt.text = port.path
						portserie.appendChild(opt)
						localStorage.setItem("com",port.path)
					}
				})
				localStorage.setItem("nb_com",ports.length)
				localStorage.setItem("com",portserie.options[1].value)
			}
			if(ports.length < nb_com){
				while(menu_opt[1]) {
					portserie.removeChild(menu_opt[1])
				}
				localStorage.setItem("com","com")
				localStorage.setItem("nb_com",ports.length)
			}
		});`
You are right in other comment that you do, which is the different folder system related to Linux version. I solved it by creating a humble script that detects if it's the first time the application is being executed and then creating an auxiliary data folder into the home user directory, so we can avoid any permission problems.
Summary: many changes are needed.  I have a first working Linux version [here, in my own github](https://github.com/agomezgar/blockly). You can download an AppImage (portable version) [here](https://github.com/agomezgar/blockly/releases/download/v1.0.0/OttoBlockly.1.0.0.AppImage)
*SPAM* I learned these things, in first place, analyzing @cparrapa work here and @fontainejp with [blocklino](https://github./fontainejp/blocklino). and secondly, creating my own graphical programming environment, [*MasayloBlockly*](https:github.com/agomezgar/masayloBlockly), which allows you to program many types of robots as well as doing general programming jobs with Arduino.

agomezgar, 2020-09-13:
> Thanks @agomezgar for such great effort and share your work
>
> Slightly similar as above, and with the dependencies suggested by gitaroktato, I have tried your linux build in my machine with Ubuntu [...]

It seems to me that the app can't communicate with your Arduino. What model are you using? It's a nano 168? Because these model is not adapted yet. I'm working on this issue.

agomezgar, 2020-09-13:
First question, really, should be: have you programmed an Arduino the usual way before you have tested this app?. Because, as I think you know, some Linux OS give some permission problems related to the DIALOUT group.

mxochicale, 2020-09-13:
> First question, really, should be: have you programmed an Arduino the usual way before you have tested this app?. Because, as I think you know, some Linux OS give some permission problems related to the DIALOUT group. If so, try opening your shell and typing:
> `sudo usermod -a -G dialout youruser `
> where **youruser** is the user name you are running the program with.

Many thanks @agomezgar 

Good news OttoDIY/blockly is now working on my Ubuntu 18.04.3 LTS x64 - Linux kernel 5.3.0-53-generic. I just need to chose Arduino Nano instead of "Arduino Nano (old bootloader)" 
[image]

Thanks for the hint on testing nano with arduino app. I did test nano with arduino app with `Board: Arduino Nano; Processor: ATMega328P; Port:/dev/ttyUSB0`. For the record, see the terminal output for the ttyUSB0 port
```
ls -l /dev/ttyUSB*
crw-rw---- 1 root dialout 188, 0 Sep 13 08:21 /dev/ttyUSB0
```

**PS.** I guess next step is to create a branch for linux_build in OttoDIY/blockly where you can add a README with some instructions for the dependencies, usage, and installations steps. Happy to review and test such PR.

agomezgar, 2020-09-13:
> 
> 
> > First question, really, should be: have you programmed an Arduino the usual way before you have tested this app?. Because, as I think you know, some Linux OS give some permission problems related to the DIALOUT group. If so, try opening your shell and typing:
> > `sudo usermod -a -G dialout youruser `
> > where **youruser** is the user name you are running the program with.
> 
> Many thanks @agomezgar
> 
> Good news OttoDIY/blockly is now working on my Ubuntu 18.04.3 LTS x64 - Linux kernel 5.3.0-53-generic. I just need to chose Arduino Nano instead of "Arduino Nano (old bootloader)"
> [image]
> 
> Thanks for the hint on testing nano with arduino app. I did test nano with arduino app with `Board: Arduino Nano; Processor: ATMega328P; Port:/dev/ttyUSB0`. For the record, see the terminal output for the ttyUSB0 port
> 
> ```
> ls -l /dev/ttyUSB*
> crw-rw---- 1 root dialout 188, 0 Sep 13 08:21 /dev/ttyUSB0
> ```
> 
> **PS.** I guess next step is to create a branch for linux_build in OttoDIY/blockly (to avoid conflict with others OS) where you can add a README with some instructions for the dependencies, usage, and installations steps. Happy to help with reviews and tests of such PR.

Those are great news!. You know? You're the first one who test the program apart from me. I am really happy that it works for you. Your suggestions, on the other hand, are welcome, and I am working on it.

mxochicale, 2020-09-13:
> > > First question, really, should be: have you programmed an Arduino the usual way before you have tested this app?. 
> > 
> > 
> > **PS.** I guess next step is to create a branch for linux_build in OttoDIY/blockly (to avoid conflict with others OS) where you can add a README with some instructions for the dependencies, usage, and installations steps. Happy to help with reviews and tests of such PR.
> 
> Those are great news!. You know? You're the first one who test the program apart from me. I am really happy that it works for you. Your suggestions, on the other hand, are welcome, and I am working on it.

Very happy to help. My suggestion is that you create a new branch called linux_build in your https://github.com/agomezgar/blockly and add detailed [instructions](https://github.com/OttoDIY/blockly/issues/38#issuecomment-691623904) for the linux_build and then you can ask for reviews to us. I am saying that because your https://github.com/OttoDIY/blockly/pull/52 in the master branch might create conflicts to the build in windows.

gitaroktato, 2020-09-13:
@agomezgar many thanks for all the help, basically everything I did was based on your work in [masayloBlocklyLinux](https://github.com/agomezgar/masayloBlocklyLinux)

I went ahead and added all the features to support Linux and Windows builds in the same repo. For this, I've separated the `compilation` folder's content in `compilation/linux`, `compilation/win32` subfolders and added helper functions in the [electronApp.js](https://github.com/gitaroktato/blockly/blob/linux_build/electronApp.js#L23)
These helper functions change paths for HTML files and executables based on OS and development/production mode.

I've added separate build commands for Linux and Windows binaries in [package.json](https://github.com/gitaroktato/blockly/blob/linux_build/package.json#L34)

What should work by now:
`npm run debug` and `npm start` runs the application in development mode, all without renaming paths manually. This should be both OK in Windows and Linux

`npm run compiler-linux` creates a Linux binary
`npm run compiler-win` creates a Windows binary

Please check my changes and let me know how I can help:
Repo URL: https://github.com/gitaroktato/blockly/blob/linux_build
Diff: https://github.com/OttoDIY/blockly/compare/master...gitaroktato:linux_build

agomezgar, 2020-09-20:
It's what i'd planned to do in the next future... I'll check the code, @gitaroktato . Thanks a lot

mxochicale, 2020-09-20:
Such a great job @agomezgar and @gitaroktato :) 

In terms of workflow, perhaps you might like to mention that users have this branch https://github.com/OttoDIY/blockly/tree/versionlinux where other people can send PRs to contribute to the documentation and test in other GNU/Linux distributions.

cparrapa, 2020-09-21:
awesome @agomezgar thanks a lot!
@mxochicale  mention where? so i think all the steps to install should be in the readme of this branch

mxochicale, 2020-11-15:
Hi  @cparrapa  

Re: 
> @mxochicale mention where? so i think all the steps to install should be in the readme of this branch

You might like to add few lines in the [README](https://github.com/OttoDIY/blockly/blob/master/README.md) of the master branch to mention that the branch of [versionlinux](https://github.com/OttoDIY/blockly/tree/versionlinux) is ready for other people to test their robot using, at least, Ubuntu 18.04 or 20.04 : )

mxochicale, 2020-11-15:
Hi  @cparrapa  @agomezgar and @gitaroktato 

I have noticed that the version of OttoBlockly for [versionlinux](https://github.com/OttoDIY/blockly/tree/versionlinux) is v1.0.0. You might like to update it to its latest version 1.3.0. Perhaps, for future versions, such as 1.4.0, you might like to include in the changelog that OttoBlockly works for both windows and Ubuntu. 

Thanks
Miguel

cparrapa, 2020-11-16:
Hi @mxochicale 

I am really lost of how it work the linux version so not sure how to do a proper readme.
Did it work for you what are the steps to install it?

No idea why is in a different version than Windows maybe is as @agomezgar established? or technically is the same 1.3.0. Blockly
We are working in 1.4.0. but there is is a lot to handle how could we for examaple reflect the changes on the linux version?

Thanks!

mxochicale, 2020-12-13:
Re: 1.3.0. Blockly mentioned above, I realised that your version 1.3.0 is showing 1.0.0: 
https://github.com/OttoDIY/blockly/blob/664059259fbbe96d3b1395f61394c1f1d08c83c0/package.json#L3

Perhaps, a minor changes has to be made to fix the appropriate version.

cparrapa, 2020-12-13:
Hi @mxochicale

You are right so we make them match versions
Thanks

mxochicale, 2021-01-10:
Hi @cparrapa 

It great to see the new version blockly with lots of nice new features https://github.com/OttoDIY/blockly/releases/tag/v1.4.0. Which makes me wonder: does blockly version 1.4.0 has been tested on its linuxversion counterpart?. If yes or not, I guess, you might like to close this issue and perhaps users can open new issues such as 
* Otto_blockly_v1.4.0 in Ubuntu18.04
* Otto_blockly_v1.4.0 in Ubuntu20.04
* Otto_blockly_v1.4.0 in Raspberry PI OS, etc

Thanks 
Miguel

cparrapa, 2021-01-10:
Hi @mxochicale 

Glad you like them, unfortunately i do not have a computer with Linux to test it, i tried to install an ubuntu to an old laptop but it turn out to be even slower after flashing.

I was hoping you can test and somehow translate the changes to the Linux branch, it should be quite smooth.

Do you think is possible, I have some  Raspberry PI also but no idea where to start?

I will close this issue then since it has been solved in this branch https://github.com/OttoDIY/blockly/tree/versionlinux

Thanks

closed 2021-01-10 by cparrapa

mxochicale, 2021-09-04:
Hi guys

just saw the amazing work of @agomezgar for the [OttoBlockly Linux Version 1.4.2. (x64)](https://github.com/agomezgar/OttoBlocklyLinux/releases/tag/1.4.2.) 

@agomezgar perhaps you would like to create a PR and update the current [documentation ](https://github.com/OttoDIY/blockly/tree/versionlinux) of your work with the new version. It is always good to have a good documentation to allow the community to replicate your work. Happy to test your last version and review your PR if that is helpful.

Thanks
Miguel