OttoDIY/blockly #40: save doesn't work on the Online Web App

sfranzyshen, 2020-07-31:
I am unable from the web app version to use the save button to save block xml files ... the open button and opening files work fine ... I'm using windows 10 and chrome Version 84.0.4147.89 (Official Build) (64-bit) ... no noticeable warnings or errors appear on the console ... even when the button is pressed ...

cparrapa, 2020-07-31:
If you have windows then better install Otto Blockly, web app is just to give a try to the UI and copy the code generated, it is not functional.

sfranzyshen, 2020-07-31:
> If you have windows then better install Otto Blockly, web app is just to give a try to the UI and copy the code generated, it is not functional.

I understand that "as is" this project isn't designed to be used from the web ... but I'm looking to adapt/modify/add-to this project to be used from a web environment ... I am expecting to encounter changes along the way with the handling of connectivity and uploading  ... i'll also have to add this issue to the list ... since it is a "key" function to have ...just thought anyone familiar with the code might know why this wasn't working ... that's all ... 

Thank You

cparrapa, 2020-07-31:
That's cool yeah lets find a way to make it work as web app, have you seen
the arduino agent?

sfranzyshen, 2020-08-01:
@cparrapa 

this problem is not related to this project's code ... it also happens in the the ["fontainejp/blocklino"](https://github.com/fontainejp/blocklino/) project  code ... but interesting it works correctly in their [web demo](https://fontainejp.github.io/blocklino.html) (different code from [main project](https://github.com/fontainejp/fontainejp.github.io) ) ... so I am comparing the differences to see if we can fix-it here ...

cparrapa, 2020-08-02:
indeed @sfranzyshen  it could be because i started this fork when Blocklino was version 1.5 after that he changed a lot adding a web demo that we do not really need but the blocks creator is really good we can start but using that factory block creator feature.

Let me know if you see other relevant difference and we update in Otto Blockly

sfranzyshen, 2020-08-02:
@cparrapa 

the code to handle the save button(s) functionality is in the electron included files only ... and are not pulled in when the web (index.html) is parsed alone. So, we have a choice ...

1) create a two versions of the code to specifically run either under electron or under the web ...
2) create one version of the code that can detect the environment it's running under and execute accordingly ...

The Blocklino project has chosen to do it the #1 approach ... but i'm going to try to accomplish #2 for the save file functions ... but once we get into the communication, building, and uploading stuff ... it might be better to split the code into two projects ... ??? I don't know about this ... my motivations are to eliminate the serial communication and compiler stages completely ... only needing to communicate over wifi and running in a interpreter environment (no compiler step) sending script over the wifi network ...

so, for now, i'm just going to concentrate on adding the save file functions when running in a browser ... to "this" code branch ... and ignore everything else (for now) ...

sfranzyshen, 2020-08-02:
> Let me know if you see other relevant difference and we update in Otto Blockly

I have seen many changes within the code ... not only between the OttoDIY Blockly and Blocklino code ... But also between Blocklino 1.8 (electron) and Blocklino 1.8 (web) code ... I've reached out to the developer for insight ... The only really big change that I see ... is the stuff for themes ...

cparrapa, 2020-08-02:
I think if you want to have all only through Wifi that is a new separately software repo that we can call ¨¨Otto Blockly Wifi¨ because if you get rid of the serial communication and compiler what happens with all the people and teachers that are using it as it is? As you said try to get as far as possible without damaging other functionalities and if happens then that means you should split into 2.

I just invite you as part of the organization so feel free to create a new repo with your ideas and test them once we are there and functional we can find ways to combine them or maybe wont be necessary at all.

I also got confused when making Otto Blockly (electron ) on what files to use i recently push all the folders and files I use to pack and still the web works but yes for the save function and the ones you suggested better to separate we can always have a link inside Otto Blockly electron installed to open the web version as alternative

If the only big change is the themes it doesn't matter i actually do not like better as simple as we have already.

Thanks happy to have you as part of the developer team!

sfranzyshen, 2020-08-04:
I have sent a pull request to address this and other problems to using blockly from the web browser ...
https://github.com/OttoDIY/blockly/pull/43

sfranzyshen, 2020-08-04:
> indeed @sfranzyshen it could be because i started this fork when Blocklino was version 1.5 after that he changed a lot adding a web demo that we do not really need but the blocks creator is really [...]

It would be nice to update the code to include the changes from the newer "block factory" from Blocklino 1.8 ... as you said it's very nice ... do you know specifically what branch and version of blocklino was used to start this project ... I can see that there were changes made to the code before the initial uploads made here ... it would help identify the changes step by step ...

cparrapa, 2020-08-04:
Great i think you solved the save function @sfranzyshen  for the blocks mode, but for the code mode still doesn't work to save "download" the .ino file.

The block factory now opens that is good and from there i can start to show you how to create the blocks.
[image]
but the thing i was referring is that if you see in this picture of recent Blocklino factory version in the top right corner there is a new button that looks like a shopping cart when i click there after designing a new block
[image]

This might help beginners to create new blocks faster, i am trying to see where was this updated. Let me know if you find it.
Thanks

sfranzyshen, 2020-08-04:
> Great i think you solved the save function @sfranzyshen for the blocks mode, but for the code mode still doesn't work to save "download" the .ino file.
> 
@cparrapa 

NOTE: The Web version ... doesn't open a "save dialog" popup like in the electron version (or like when opening files) ... it will just save the file directly to the default download location without a prompt ... check your download folder to see if the files are there ... or from the browser check the "downloads" from the pull-down menu ...

If the files are not there ... What browser version and operating system are you using? because from windows 10 64bit chrome I'm able to save the ".bloc" files and the code files too ... both ".ino" and ".py" files ... and I can do it from both icons ... the "save" button on the main code view screen ... and the one from the pop-out code viewer on the right side of the screen ...

[image]
[image]

Could you test saving code files from the web version of the original blocklino project page ... and see if you are able to save code files from there?

https://fontainejp.github.io/blocklino.html

Thanks

cparrapa, 2020-08-04:
Hi @sfranzyshen 
strange before i was not able to download .ino files not it works, never mind now is perfect great job!

sfranzyshen, 2020-08-04:
> but the thing i was referring is that if you see in this picture of recent Blocklino factory version in the top right corner there is a new button that looks like a shopping cart when i click there after designing a new block
> It creates an actual block! with the code and libraries embedded
> Now i don't know if the block disappear after deleted or close the program but this might help beginners to create new blocks faster, i am trying to see where was this updated. Let me know if you find it.

@cparrapa 
The version of the blocklino that I have been working from is 1.8 (current) ... from your pictures I'm not sure what version that is ... as it seems to have changed yet again ... and there are big differences between the Electron version (1.8) and the Web version (1.8) that I have not received a response back from the developer ... i'm going to try to open an issue on github ...

Blocklino 1.8 (Web Browser)
[image]

Blocklino 1.8 (Electron)
[image]

The electron version allows directly adding blocks back to blockly ... but the web version only allows saving to ".bf" block factory files ... so the more I look at the changes between the electron and the web versions ... the more I see why the split in code bases ... this maybe what needs to happen here also ... we may need to split the code into two projects ... one for electron ... and one for web.

sfranzyshen, 2020-08-04:
> Hi @sfranzyshen
> strange before i was not able to download .ino files now it works, never mind now is perfect great job!

aahhh ... it was cached ... if you had done a shift+reload (holding down the shift key while reloading the page) it would have pulled the newly changed files ... other wise it would have pulled from the cached files ... not using the updated file. but it would eventually ... pull the new files ... if you had closed your browser ... or restarted your machine ... or the cache simply expired ... that's my guess 💯

cparrapa, 2020-08-04:
Hi @sfranzyshen
I see that actually 1.8 version of factory looks better with more ways to save the blocks i saw the 1.6 version we can try the latest and see how can we design new blocks with it.

sfranzyshen, 2020-08-04:
> Hi @sfranzyshen
> I see that actually 1.8 version of factory looks better with more ways to save the blocks i saw the 1.6 version we can try the latest and see how can we design new blocks with it.

I'm going to go ahead and close this issue since it has been handled ... and open a separate new issue for updating the block factory code to the 1.8 version ...

closed 2020-08-04 by sfranzyshen