OttoDIY/blockly #41: JavaScript Generator Support

sfranzyshen, 2020-07-31:
I am attempting to combine together OttoDIYESP(LIB) and [Elk ](https://github.com/cesanta/elk)(a tiny embedded JavaScript engine that implements a small, but usable subset of ES6,) resulting in a JavaScript-able OttoDIY envirnoment. My work is being done for the esp8266 platform because of the arduino nano ram limits ... and would like to use this project as part of the web interface ...

### **I would like to add javascript code view and output to this blockly project.** 

I have only glanced over this code ... but I know in the regular blockly javascript (ES5) there is supported so I am hoping anyone who is familiar with this code might be able to point me in the right direction ... using the javascript generators from the regular blockly won't work or would need to be changed to support the limited subset of the Elk javascript interpreter ... so staring from scratch would probably be best ... in my work the OttoDIY API is exposed almost exactly like the C++ Arduino environment within the javascript environment

any blockly guru's want to help me workout the generator stuff within this project ???

cparrapa, 2020-07-31:
Feel free to experiment with the ESP repo is there for experimetation, I am
not a guru in Java but i can help you with the blocks

cparrapa, 2020-08-02:
So this javascript code also need to convert from C++?

sfranzyshen, 2020-08-02:
> So this javascript code also need to convert from C++?

@cparrapa yes ... all blocks would need to have code created for generating JavaScript code (completely new generator) ... the one difference between writing code for python and JavaScript ... is I already have a path for porting OttoDIY API into the JavaScript environment ... and the structure is very close to the original C++ so porting to JavaScript, compared to python, will be a snap. But, over time I'd like to see a full port of the OttoDIY API into python ... So while I'm porting code for JavaScript, I can start to add the code for python ... but it would be putting the cart before the horse ... without a python OttoDIY library ...

cparrapa, 2020-08-02:
Yes we need the libraries first also for Javascript?

sfranzyshen, 2020-08-02:
> Yes we need the libraries first also for Javascript?

I'm using Elk JavaScript Engine as a arduino library ... combined with the OttoPLUS_V9.ino ... I am able to expose the OttoDIY API into the JavaScript Virtual Machine ... Allowing for JavaScript Scripts to be created that can call the OttoDIY API exactly the same way that is done from the C++ code ... C and JavaScript have a very similar structure and this works well ... there are obvious differences and creating/changing code for blockly will still need to be done ... the only big difference ... there is no need to compile the code just send it up to the controller over wifi ... So unlike python ... I already have the OttoDIY API ported to JavaScript (in a sense ... really a wrapper) ... since I'm just reusing the existing C++ code. Unless the OttoDIY API C++ code can be added to micropython firmware somehow ... all of the OttoDIY API code would (will) need to be reproduced within the micropython environment.

Since my goal is to have a script-able (no compiler step) "out-of-the box" (no development software to install) all web browser based Otto ... The Elk JavaScript environment is a perfect fit ... but i'm also eyeing moving to the ESP32 and a more robust JavaScript engine ... so many options ...

cparrapa, 2020-08-03:
Sounds interesting if we have in java you think we can integrate processing applications and do graphical interactions with Otto?

sfranzyshen, 2020-08-03:
> Sounds interesting if we have in java you think we can integrate processing applications and do graphical interactions with Otto?

JavaScript and Java are completely different things ... and processing would never directly run from the micro controller ... but any processing app (running on a laptop for example) would be able to communicate to otto over the network ... as would anything running in the browser ... so ... I could see a processing app running on a laptop ... interacting with otto ... or a browser based web app (served up from otto it'self) that could also interact with otto ... we can modify or fork the bluetooth firmware to accept commands over the network ... and control multiple otto robots simultaneously from a hosting control program running from a laptop ... this however is not related to the JavaScript otto project ... or adding JavaScript support to the otto blockly project ... however, it could be done in the core of OttoDIY API (LIB) it'self ... perhaps adding it to the OttoDIYESP project ... This is within my plans ... as I too need this within the scope of my goals here ...

cparrapa, 2020-08-03:
got it sounds good!

sfranzyshen, 2020-08-17:
while looking at what it will take to port the OttoDIY API (Otto.h) to Python ... I've decided to give a JavaScript port a shot too ... A full port to the [Espruino JavaScript Environment](https://github.com/espruino/Espruino) ... In the [planning stages right now](https://github.com/sfranzyshen/OttoDIYJavaScript) ... I'm not abandoning my plans with Elk ... Just experimenting with a more complete JavaScript Environment ...

cparrapa, 2020-08-18:
Looks good i saw Espruino before in other robot but never explored worth checking out

sfranzyshen, 2020-09-13:
work has begun on [porting the OttoDIY API to the espruino javascript environment](https://github.com/sfranzyshen/OttoDIYJavaScript) ... as with the python port ... anyone with javascript experience is encouraged to participate ...

sfranzyshen, 2020-09-22:
The JavaScript threaded model conflicts with the OttoDIY blocking design ... I have fallen back from this work for right now and going back to the beginning with the original Cpp code. [I'm refactoring the original code to be non-blocking and universally cross-platform for all Arduino environments](https://github.com/sfranzyshen/OttoDIYArduino) ... I'll then propagate the changes to the python and (finally) the javascript platforms ...

sfranzyshen, 2020-09-29:
because of JavaScript's asynchronous design conflicts with the nature of the OttoDIY step-by-step code flow we have started a new project ([OttoDIYAsync](https://github.com/sfranzyshen/OttoDIYAsync)) to split the original Arduino code into a standard 'blocking' platform and for more advanced students a 'non-blocking platform' ... once this conversion has happened we will resume development on the JavaScript version ...