OttoDIY/blockly #42: python code view and output do not work

sfranzyshen, 2020-07-31:
In either the web or electron version no matter what python board selected ... neither the code view or code save result in python code being generated ...

cparrapa, 2020-07-31:
Hi Scott
Good point we need to create the phyton one if you have the I can add into
the blocks for you

sfranzyshen, 2020-07-31:
I'm confused by your response ... 

1) I open blockly and select the esp8266-python board from the pull down menu
2) I than select an example "dance" from the popup ... it open and displays the blocks fine
3) when I click on the block/code view toggle button ... no code appears in the window (the python code generator isn't working)

there are no problems when selecting arduino nano-cpp board with generating the cpp code ...

I am missing something?

sfranzyshen, 2020-08-01:
ok, I can see that the blocks that were added to the blocklino base code (which looks to be the base code used for building this project) in order to support the otto features .... don't have the supporting code needed to generate the python code ... but the blocks that were already in blocklino do ...

are there any plans to complete this at some point? I was hoping to use the comparison between the cpp and python code generators to help create a javascript generator ... I'm still trying to understand how all these files add up to actually generate the code ... any insight would be greatly appreciated

cparrapa, 2020-08-01:
Hi @sfranzyshen 

1. Yes we have ESP8266 Python but the generator is has not been made for blockls of Otto. How to convert C++ to Python? this 2. For C++ aka Arduino code no problem i did that
3. It is not that is not working there is nothing.

There was an idea to do this, but not a plan or a clear path i do not know where to start to do this as i am new to Python it looks easier but i guess we have to make Otto work in Python first no? it seems you have more experience with Python your input will be very valuable thanks.

Like with the other issue if you make the code for Python, i can do the blocks generator. I am also working on a block creator similar to blocklino that can make this easier for you but meanwhile [you should divide the code in this table](https://docs.google.com/spreadsheets/d/1E_wf6ls3sdqTvQDtcUtNwovEI5ymoyx0vgsi4uC-0f8/edit?usp=drive_web&ouid=115894788587809324360) that is the very first step.

sfranzyshen, 2020-08-01:
> Like with the other issue if you make the code for Python, i can do the blocks generator. I am also working on a block creator similar to blocklino that can make this easier for you but meanwhile [...]

@cparrapa Thank you for this link ... it is the first time I have seen this type of breakdown ... however, the file layout (what goes in to what file) is still not clear to me ...

### **also, how complete is this list ? ... are all of the blocks that appear in the OttoDIY/blockly project represented here in this list ?** 

I am willing to work on the python code ... although my main motivation is to add [JavaScript support](https://github.com/OttoDIY/blockly/issues/41) ... but I'm willing to  work on both simultaneously ... as you have pointed out ... the OttoDIY API needs to first be ported to python ... i'd also like to see a [full port of the OttoDIY API](https://github.com/sfranzyshen/OttoDIY_Python) to the micropython platform ... but this would require a major change to the OttoDIY project ...  The hardware (nano & breakout board) would need to be change and the code needs to be ported to python ... so, this is a long term wish for me ... in the meantime i'm willing to work on the python stuff while I'm working on the [JavaScript ](https://github.com/sfranzyshen/OttoDIY_JavaScript)goal

cparrapa, 2020-08-02:
Hi @sfranzyshen 

I honestly did not test python generator before but is in the roadmap.
The list is just the main ones to give you an idea, i can teach you how it works in a video call just reach out to me in hangouts or messenger.

All your ideas sound great and we can slowly get there thanks for your support.

sfranzyshen, 2020-08-02:
> The list is just the main ones to give you an idea, i can teach you how it works in a video call just reach out to me in hangouts or messenger.
> 

I'm still getting familiar with the code ... once I have a good feel for things ... then I will definitely take you up on your offer for a video call ... Thank You for your feed back ... I can see a light at the end of the tunnel ...

sfranzyshen, 2020-08-12:
To make any progress on the OttoDIY Python Port the ... Servo & Oscillator libraries need to be ported first. The micropython documentation has an example that demonstrates manipulating the servo [using the PWM module](http://docs.micropython.org/en/latest/esp8266/tutorial/pwm.html?highlight=servo#control-a-hobby-servo) ... and thesheep has a [servo python module](https://bitbucket.org/thesheep/micropython-servo/src/tip/servo.py) that could be extended to match the Arduino Servo driver format ... Or the Oscillator Code could be modified to handle the servos directly (Consume the Servo Code) ... I think the rest of the Oscillator Code can be ported relatively easily ...

... If there is anyone with strong python skills that could help get us started ... I'm willing to assist ...

cparrapa, 2020-08-12:
great find i am also new to python so we will see if in the community someone appear. But maybe there is a shortcut, i just remember a copycat of Otto from a Chinese company that turned not successful but their code was in python unfortunately they did not made open source but i manage to dig in and found this http://wiki.labplus.cn/index.php?title=Bot:bit

I can see a lot of things similar to our libraries there but never found a way to test, what board can we use for this @sfranzyshen ?

sfranzyshen, 2020-08-13:
> i just remember a copycat of Otto from a Chinese company that turned not successful but their code was in python unfortunately they did not made open source but i manage to dig in and found this http://wiki.labplus.cn/index.php?title=Bot:bit

this is built off of the micro:bit hardware and micropython ... seems to drive the servos via a i2c interface ... some useful snippets ... but not complete enough ...
 
> I can see a lot of things similar to our libraries there but never found a way to test, what board can we use for this @sfranzyshen ?

I have A NodeMCU Esp8266 with base micropython firmware ... this is what I'll be working from ... I have a ESP32 board that I'll also be targeting ... but any platform that supports micropython should be able to adapt the code created ...

cparrapa, 2020-08-13:
I also have a couple of NodeMCU Esp8266 but how to know if they have micropython firmware

cparrapa, 2020-08-13:
I am getting a bit lost since i am a beginner i this python world but i found another robots that already sorted that out https://www.beyond-coding.org.tw/component/sppagebuilder/?view=page&id=89 but with micro:bit

sfranzyshen, 2020-08-14:
@cparrapa 

When I reference "Porting OttoDIY API" to python ... I am referring to everything within the "Otto9.h" file ... Yes, someone is able to develop for OttoDIY without using "Otto9.h" ... coding everything out ... Just Like the "Otto_smooth_criminal_V9.ino" example demonstrates ... And The same can be done within the micropython environment ... we can either create a "otto9.py" library for micropython (this is what I'm working toward) ... otherwise we will need every example to reproduce the core functions that are in "Otto9.h" within each of the students code just like what is done in the "Otto_smooth_criminal_V9.ino" code ... personally ... I'm moving away from any code that isn't built using "Otto9.h" ... or I'm porting it to use "Otto9.h" ...  

my intent for "OttoDIY python support" currently is to create "servo.py" and "oscillator.py". And in time I'll work on the other libraries ... working toward a "otto9.py" python library ... than I'll port the examples to python using the otto9 python library ... and finally the goal of adding the needed code to blockly to generate code for the blockly otto blocks ... into python code ... 

Also, I was unable to find any of the code or software for the link in your last post above ...

cparrapa, 2020-08-14:
It makes total sense what you said now i understand more but i am still not able to help much with the porting itself, i need to learn first. Here i found the extension repo for that Otto remix robot with microbit https://github.com/beyond-coding-tw/pxt-nexusbot

This is another one that i found https://github.com/aguegu/pxt-sourcekit-otto
it is from a friend developed some python programs for calibration even a pcb for it but as you said it was a way of going around the Otto9.h original libraries, unfortunately he is busy now to help.
[otto_gui.zip](https://github.com/OttoDIY/blockly/files/5077002/otto_gui.zip)

I hope is helpful, meanwhile we can make a call to help in the Otto Builders group. How should we post it?

sfranzyshen, 2020-08-14:
The other projects on their own ... work ... however, I do not see them fulfilling the goals of having OttoDIY within a micropython environment ... nor supporting "otto blocks" python code generation from blockly ...
 
> I hope is helpful, meanwhile we can make a call to help in the Otto Builders group. How should we post it?

A call for python developers to assist in the porting of the OttoDIY API (Otto9.h) to the micropython environment on the ESP8266 & ESP32 platforms ... Either start an official project for the python port ... or you can direct them over to https://github.com/sfranzyshen/OttoDIY_Python/issues

Peace 😃

cparrapa, 2020-08-15:
HI @sfranzyshen these projects my idea was to take their code and use into our Blockly some of them already solved the movements, especially PLENbit

The call looks great but is it possible to put as a main repo for https://github.com/OttoDIY/ ? in this way i think more people will see it and they will be willing to contribute. I sent you invitation weeks ago to be part of the organization so that you can create all this repos there

thanks.

sfranzyshen, 2020-08-15:
> The call looks great but is it possible to put as a main repo for https://github.com/OttoDIY/ ? in this way i think more people will see it and they will be willing to contribute. I sent you invitation weeks ago to be part of the organization so that you can create all this repos there

I tried to transfer the project to OttoDIY but was unable to create or transfer public repositories  ... I get the following message ...

"You don’t have the permission to create public repositories on OttoDIY"

☹️

cparrapa, 2020-08-16:
@sfranzyshen now worries just sent again try and let me know if you are able.

sfranzyshen, 2020-08-16:
I transferred the python project to OttoDIY ...

https://github.com/OttoDIY/OttoDIYPython

cparrapa, 2020-08-16:
ok great posted on social media and community groups, now i remember @kevinmcaleer is very advanced in Python we talked about this before, he is busy with other projects but we can have a call with him to learn where and how to start https://github.com/kevinmcaleer/SMARSLab

sfranzyshen, 2020-08-30:
https://github.com/OttoDIY/OttoDIYPython/issues/1#issuecomment-683372877
https://github.com/OttoDIY/OttoDIYPython/issues/2#issuecomment-683373210

sfranzyshen, 2020-09-02:
I am excited to feedback that we now have some working python code ... The Servo, Oscillator, and Motion Code is now ported and working (from what I can tell ... I haven't actually put it into a OttoDIY yet) Here is what the test code looks like ... [Otto_allmoves_V9.py](https://github.com/OttoDIY/OttoDIYPython/blob/master/Otto_allmoves_V9.py)

```
"""
Otto All moves python test 
OttDIY Python Project, 2020 | sfranzyshen
"""
import otto9, time

Otto = otto9.Otto9()
Otto.init(5, 12, 13, 14, True, 0, 1, 2, 3)
Otto.home()

Otto.walk(2, 1000, 1)			#-- 2 steps, "TIME". IF HIGHER THE VALUE THEN SLOWER (from 600 to 1400), 1 FORWARD
Otto.walk(2, 1000, -1)			#-- 2 steps, T, -1 BACKWARD 
Otto.turn(2, 1000, 1)			#-- 3 steps turning LEFT
Otto.home()
time.sleep_ms(100)  
Otto.turn(2, 1000, -1)			#-- 3 steps turning RIGHT 
Otto.bend(1, 500, 1)			#-- usually steps =1, T=2000
Otto.bend(1, 2000, -1)     
Otto.shakeLeg(1, 1500, 1)
Otto.home()
time.sleep_ms(100)
Otto.shakeLeg(1, 2000, -1)
Otto.moonwalker(3, 1000, 25, 1)	#-- LEFT
Otto.moonwalker(3, 1000, 25,-1)	#-- RIGHT  
Otto.crusaito(2, 1000, 20, 1)
Otto.crusaito(2, 1000, 20, -1)
time.sleep_ms(100)
Otto.flapping(2, 1000, 20, 1)
Otto.flapping(2, 1000, 20, -1)
time.sleep_ms(100)
Otto.swing(2, 1000, 20)
Otto.tiptoeSwing(2, 1000, 20)
Otto.jitter(2, 1000, 20)			#-- (small T)
Otto.updown(2, 1500, 20)		#-- 20 = H "HEIGHT of movement"T 
Otto.ascendingTurn(2, 1000, 50)
Otto.jump(1, 2000)

Otto.home()
```

This is ONLY the motion code ... sensors, sound, matrix, and gestures still all need to be done ...
https://github.com/OttoDIY/OttoDIYPython/issues

cparrapa, 2020-09-02:
wow this is amazing i love how simple the python code looks like, this is gonna be big!
thanks

sfranzyshen, 2020-09-22:
Thanks to @mishafarms ... [we are getting even closer to our goal](https://github.com/OttoDIY/OttoDIYPython/pull/14) ... close enough to start working on addressing "This Issue" by adding the python code to the blockly code (python) generator 

@cparrapa I think we have enough to start adding code ... I still have not looked closely at the blockly code enough to know where to start ... I was hoping you maybe able to help?

It's not clear to me ... but I do not think the blockly (blocklino) electron app is working with uploading python files to the board ... of course saving the file and manually uploading to the board works ... but it would be nice to get this working too ... I'll look again at the latest blocklino and see if they have it functioning ...

cparrapa, 2020-09-22:
Hi @sfranzyshen 
great news, yes i can help but i don't have a python board to do the tests and we have to find a way to upload codes like we do for  Arduino

sfranzyshen, 2020-09-22:
> great news, yes i can help but i don't have a python board to do the tests and we have to find a way to upload codes like we do for Arduino

Almost Any Esp8266 or ESP32 development board with 4MB flash should do ... like a nodemcu or wemos ... but that's not actually needed to get our code into blockly ... we can see the resulting code generated from within blockly and verify it elsewhere ... you sent me a [link to a spreadsheet](https://docs.google.com/spreadsheets/d/1E_wf6ls3sdqTvQDtcUtNwovEI5ymoyx0vgsi4uC-0f8/edit#gid=0) showing some (not all) of the blocks and code for the Cpp generator ... we now have the structure and python code snippets to the necessary format outlined from the spreadsheet ... but where does this actually get edited into the blockly code generators ??

cparrapa, 2020-09-22:
Ah ok i have not done for python but i guess the process is the same if you see here https://github.com/OttoDIY/blockly/blob/e866e2e8602eaa2dc98ee427f0859606239523c3/www/blocs%26generateurs/actionneur.js#L21 there is an example of how it was already made for other blocks.
Blockly.Blocks is the actual graphical representation of the block.
Blockly.Arduino is the generator for C++
then Blockly.Python must be to generate in Python 
for all blocks they are all here https://github.com/OttoDIY/blockly/tree/master/www/blocs%26generateurs in Otto i think we only have for Arduino so far so i guess just need to add new lines here https://github.com/OttoDIY/blockly/blob/master/www/blocs%26generateurs/otto.js

mishafarms, 2020-09-22:
I have the beginning of the blackly code for Otto in mishafarms/edublocks-micropython.
I am trying to use the same blockly code you provided. I had to modify a little because edublocks-micropython isn’t using closure and I have not been able to figure out how to include the msg files. But the generators should work. This code runs on the Otto and allows us to program with nothing more then a browser.

If you want a demo, I can configure a system to be available on the internet.

Michael

mishafarms, 2020-09-22:
I am willing to port into the Otto blockly, I just wanted to get more done first and then I would like to put it under the OttoDIY repositories.

cparrapa, 2020-09-23:
Hi @ mishafarms 

This looks amazing, yes please port the python generators that would be a great progress indeed.
Feel free put  edublocks-micropython  under the OttoDIY repositories. you are part of the organization.
Let me know what else i can help with
Thanks!

sfranzyshen, 2020-09-24:
> gitlab.playcafe.us/edu.html
@mishafarms this link times out ...

sfranzyshen, 2020-09-24:
> I have the beginning of the blockly code for Otto in mishafarms/edublocks-micropython. This will be blockly on the Otto itself. No need to upload code. 

@mishafarms what is the backend used to serve up files and respond to the ws calls? ... does it use the web-repl ??
update: This?? https://github.com/mishafarms/esp32-micropython

mishafarms, 2020-09-24:
@sfranzyshen Contact me directly and I will see what the issue is with the link.

The backend does use web-repl. It also starts it's own HTTP server yes using [https://github.com/mishafarms/esp32-micropython](url). I have not been able to get some other micropythons working specifically a newer release with PSRAM. But I will play with that once I have everything else working.
The blocks are a little different from the ones in OttoDIY blockly. I will talk to you about the differences and see about getting them the same. 
I am almost done with the basic Otto python code. I am going to add OttoHumanoid code even though we don't yet have support for it. 
Michael

sfranzyshen, 2020-09-24:
@mishafarms 

> The blocks are a little different from the ones in OttoDIY blockly. I will talk to you about the differences and see about getting them the same.

I also think the blocks should be redone ... especially exposing the _moveServos(),  oscillateServos(), and  _execute() functions so that advanced students can build custom moves ...

> I am almost done with the basic Otto python code. I am going to add OttoHumanoid code even though we don't yet have support for it.

why do we not have support for OttoHumanoid ? I thought you had 6 servo connectors on the pcb just four labeled ... I have also been refactoring [the original Cpp Arduino code to be universally cross platform across all arduino platforms](https://github.com/sfranzyshen/OttoDIYArduino) ... I am merging the humanoid code in too ... working toward a single code set for both the PLUS and Humanoid Otto's ...

~peace

mishafarms, 2020-09-24:
@sfranzyshen we will support the OttoHumanoid.

Michael

sfranzyshen, 2020-09-27:
### We Now Have Humanoid Support

even more progress has taken place over at the[ OttoDIYPython project](https://github.com/OttoDIY/OttoDIYPython) ... we now have added the humanoid support 😄 ... this is taken from the work done over at the [OttoDIYArduino project](https://github.com/sfranzyshen/OttoDIYArduino) that is refactoring the OttoDIY Robot API (Otto9.h & Otto9Humanoid.h) code to be universally cross-platform (all-in-one) and 100% backwards compatible for all Arduino environments ... Arduino Library