OttoDIY/blockly #45: update the block factory code to the blocklino 1.8 version ...

sfranzyshen, 2020-08-04:
update the block factory code to the newer blocklino 1.8 version ...

this is a continuation of the discussion brought up from the previous issue 

https://github.com/OttoDIY/blockly/issues/40 save doesn't work on the Online Web App

cparrapa, 2020-08-04:
makes sense better separate

sfranzyshen, 2020-08-05:
I have added the blocklino 1.8 block factory to OttoDIY Blockly Web version ... I'm still attempting to work out the electron version code ... but there is quite a bit of code that has changed with the electron code that an easy swap isn't possible ... I will push the changes to my fork ... but I'll wait to make a pull request until I work out the problems with the electron code ...
https://github.com/sfranzyshen/OttoDIY_blockly

cparrapa, 2020-08-05:
It is strange that is in French again i translated all of that to English

sfranzyshen, 2020-08-05:
after reviewing the changes that have been made in blocklino 1.8 for the block factory electron version ... my conclusion is that it would require too much work to merge both the electron version and the web version into a unified code set ... so my solution is to split from this main blockly project ... and focus working on the website only version within that spit ... I plan to work toward updating the blocklino 1.8 block factory electron version into this OttoDIY blockly project at some point in the future ... so i'll leave this issue open for now ...

Splitting this project for web only version to OttoDIY_blockly_web https://github.com/sfranzyshen/OttoDIY_blockly_web

sfranzyshen, 2020-08-05:
> It is strange that is in French again i translated all of that to English

the block factory code does not use the language files ... it hard codes in labels directly in the source code. I did not take the time to rework your files (the changes were too great!) but I rather replace them with the newer code from the blocklino 1.8 block factory ... I am going to also translate the newer code ... but i'm going to add in the language support ... just like in the main blockly code.

cparrapa, 2020-08-06:
Yes now i remember i had to manually modify the French words. let me know if any help is needed 
thanks!

sfranzyshen, 2020-08-06:
> Yes now i remember i had to manually modify the French words. let me know if any help is needed
> thanks!

Thankfully, You have already done most of the work ... I can use the comparison between your files and the new files to find the translations ... and then just cut and paste 😄 ... but using the lang/code.js method ...

closed 2021-10-30 by cparrapa