OttoDIY/blockly #46: Using arduino_web_server for web version of otto blockly in Ubuntu 18.04x64

mxochicale, 2020-08-15:
Dear @cparrapa et al. 

Related to https://github.com/OttoDIY/blockly/issues/38, one potential solution might be the use of the web version of Otto blockly by running a python arduino_web_server. That said, I made a [PR#116](
https://github.com/BlocklyDuino/BlocklyDuino/pull/116) adding few instructions on the use of BlocklyDuino in Ubuntu 18.04x64. Then, it is just came to my mind the idea of trying such similar approach as shown below

[image]

However, I am a bit stuck here when trying to related the upload functions of both `blocklyduino/index.html` and `/www/index.html`: 
https://github.com/BlocklyDuino/BlocklyDuino/blob/gh-pages/blockly/apps/blocklyduino/index.html#L300
```
              <button type="button" onclick="uploadClick()">Upload</button>
```
https://github.com/OttoDIY/blockly/blob/master/www/index.html#L131
```
		<button id="btn_flash" class='btn btn-arduino' data-toggle="modal" data-target="#message">
			<span class="fa fa-arrow-right fa-1_5x fa-fw"> </span>
		</button>
```

What do you think? Any ideas?

cparrapa, 2020-08-16:
Hi @mxochicale

Good to see you here, so if i understand well your idea is to run the web version in an offline way? or online but how to solve the webusb? 
Yes we are also trying to fix the web version but the arduino builder now CLI needs some work to solve the uploading.

Thanks for the PR i am not sure how to test this i dont have ubuntu pc does this solution could work for Win and Mac too?

mxochicale, 2020-08-16:
Thanks Camilo, see some of my replies to your questions.

> Good to see you here, so if i understand well your idea is to run the web version in an offline way? or online but how to solve the webusb?

BlocklyDuino is running in an online way using python arduino_web_server in Ubuntu 18.04x64 (see [PR#116](
https://github.com/BlocklyDuino/BlocklyDuino/pull/116)  for details).

> Yes we are also trying to fix the web version but the arduino builder now CLI needs some work to solve the uploading.

Good luck! Let me know how this is going on! I am particularly interested with the use of Otto Blockly for machines under GNU/Linux OS.

> Thanks could this work for Win and Mac too?

In the [README](https://github.com/BlocklyDuino/BlocklyDuino/blob/gh-pages/README.md) of BlocklyDuino states the following "Arduino board on Windows, Mac OS X and Linux systems." I imagine that works for Windows and Mac.

> We can always try this approach too here if it doesnt affect the rest of the code

The current approach might be a temporal one. I will be updating this issue as I am not sure, at this moment, that a PR is ready for a cross-platform of webusb for Otto Blockly.

cparrapa, 2020-08-17:
Hi @mxochicale 
This is great i think it is worth making it official now
How can i include your PR#116 also here?

mxochicale, 2020-09-12:
I am closing this one as this is appropriately followed up in https://github.com/OttoDIY/blockly/issues/38

mxochicale, 2020-09-12:
I am closing this one as this is appropriately followed up in #38

closed 2020-09-12 by mxochicale