OttoDIY/blockly #47: El botón de la boca para dibujar en la matriz LED 8x8 no parece funcionar.

agomezgar, 2020-08-16:
Hola, chicos. Como ya sabéis alguno, estoy trabajando en mi propio proyecto, [MasayloBlockly](https://github.com/agomezgar/masayloBlockly), y he aprendido mucho estudiando este trabajo. El caso es que creo haber descubierto un error en el botón que permite dibujar en la matriz 8x8 que hace de boca (nombre de botón otto9_matrix8x8). Como no domino mucho aún la tecnología GIT, os paso por aquí el código que he modificado en el archivo /www/blocks&generateurs/otto.js a partir de la línea 1256:
`Blockly.Blocks["otto9_matrix8x8"] = {  init: function() {
  this.appendDummyInput().appendField('  ').appendField(' 0').appendField(' 1').appendField(' 2').appendField('  3').appendField('  4').appendField(' 5').appendField(' 6').appendField(' 7');
  Blockly.FieldCheckbox.CHECK_CHAR= '▉'
  this.appendDummyInput().appendField('0 ')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel0')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel1')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel2')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel3')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel4')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel5')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel6')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel7');
 this.appendDummyInput().appendField('1 ')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel8')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel9')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel10')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel11')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel12')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel13')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel14')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel15');
 this.appendDummyInput().appendField('2 ')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel16')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel17')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel18')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel19')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel20')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel21')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel22')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel23');
 this.appendDummyInput().appendField('3 ')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel24')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel25')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel26')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel27')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel28')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel29')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel30')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel31');
 this.appendDummyInput().appendField('4 ')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel32')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel33')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel34')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel35')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel36')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel37')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel38')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel39');
 this.appendDummyInput().appendField('5 ')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel40')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel41')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel42')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel43')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel44')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel45')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel46')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel47');
 this.appendDummyInput().appendField('6 ')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel48')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel49')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel50')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel51')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel52')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel53')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel54')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel55');
 this.appendDummyInput().appendField('7 ')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel56')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel57')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel58')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel59')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel60')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel61')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel62')
    .appendField(new Blockly.FieldCheckbox("FALSE"), 'Pixel63');
 this.setInputsInline(false);
 this.setPreviousStatement(true, null);
 this.setNextStatement(true, null);
 this.setColour("#59646f");
 this.setTooltip('');
 this.setHelpUrl("https://learn.adafruit.com/adafruit-neopixel-uberguide/arduino-library-use");
},
};
Blockly.Arduino.otto9_matrix8x8 = function() {
  Blockly.Arduino.definitions_['otto9_matrix_def'] = '#define DIN_PIN A3\n'
  + '#define CS_PIN A2\n'
  + '#define CLK_PIN A1\n'
  + '#define LED_DIRECTION 1';
  Blockly.Arduino.setups_['otto9_matrix']='Otto.initMATRIX( DIN_PIN, CS_PIN, CLK_PIN, LED_DIRECTION);';
var code = '';
for (var i=0; i<64; i++) {

 if (this.getFieldValue('Pixel' + i) != 'rgb(255, 255, 255)') {
     var on = this.getFieldValue('Pixel' + i)== "TRUE"? "1" : "0";
     var row= i +1
     {if  (i >= 0 && i <= 7)row=0}{if  (i >= 8 && i < 16)row=1}{if  (i >= 16 && i < 24)row=2}{if  (i >= 24 && i < 32)row=3}
     {if  (i >= 32 && i < 40)row=4}{if  (i >= 40 && i < 48)row=5}{if  (i >= 48 && i < 56)row=6}{if  (i >= 56 && i < 64)row=7}
     var col= i
     {if  (i > 1 && i <= 7)col=i}{if  (i >= 8 && i < 16)col=i-8}{if  (i >= 16 && i < 24)col=i-16}{if  (i >= 24 && i < 32)col=i-24}
     {if  (i >= 32 && i < 40)col=i-32}{if  (i >= 40 && i < 48)col=i-40}{if  (i >= 48 && i < 56)col=i-48}{if  (i >= 56 && i < 64)col=i-56}
     code += ' Otto.setLed('+row+','+col+',' + on + ');\n'
 }
};
for (var i=0; i<8; i++) {if (this.getFieldValue('eyes_pixel' + i) == 'TRUE')row = 0;};
return code;
};
`

cparrapa, 2020-08-16:
gracias @agomezgar supongo que podemos copiar tu otto.js alguna otra modificación?

agomezgar, 2020-08-16:
Ninguna de momento. Recuerda que apenas estoy empezando a aprender sobre
todo este mundo.

cparrapa, 2020-09-28:
thanks again @agomezgar 
fixed in this commit https://github.com/OttoDIY/blockly/commit/643c001255d80d1c21aa1dcac54b42b642225780

closed 2020-09-28 by cparrapa