OttoDIY/blockly #48: Bug in setting menu "Language & Blocks"

cparrapa, 2020-08-17:
Even after I select beginner then click accept and then back into this menu log, it still shows advance.

sfranzyshen, 2020-08-17:
In the electron version, within the settings popup window, the "Level Dropdown"  defaults to "Advanced" and in the web version it defaults to "Maker" ... either way the status of the "Level Dropdown" is not being stored ... only the checkbox items below are stored ... so the next time the settings popup window is opened it always uses the default for the dropdown but the checkbox items below are stored and work correctly ... by the way this does work correctly within blocklino 1.8 ... I'll try and see what changed ... 

either way ... it does work ... sort of ...

cparrapa, 2020-08-18:
So this is happening because i put an refresh after click the accept button but i had to do this so people no need to press F5 every time they want to change the language, so the auto refresh is bringing back the "Advanced" toolbar too.

I see that in blocklino 1.8 it is not solved completely because with language is coming back to French all the time.

cparrapa, 2021-01-08:
there must be a way to make it refresh when changing language but not when changing level so that the blocks are different per toolbar

cparrapa, 2021-06-06:
This has been fixed by @logix5 here https://github.com/OttoDIY/blockly/commit/013c60ac8680f95b9b200eaafdd8ab17081f8eb7

closed 2021-06-06 by cparrapa