OttoDIY/blockly #58: I want to update the otto blockly

logix5, 2020-10-13:
I want to update the otto blocky to add more block with other sensors.  My question is simple, because when i download the source code and execute the index.html the blocks of the code don´t appear.  The code to show the blocks are the toolboxs folder, but this code i need to know hot to show.

Please let me know thanks

[image]

cparrapa, 2020-10-14:
Hi @logix5 
You can do a Pull request of the new sensors or blocks you have created and we help you to add in the bar for the next release
thanks

agomezgar, 2020-10-22:
I see you are always busy, @cparrapa. @logix5 , you must investigate about Electron technology. I started, as you, analyzing @cparrapa 's work and I ended creating [my own app](https://github.com/agomezgar/masayloBlockly). I suggest you start by reading[ this link.](https://www.electronjs.org/). You will discover that we are not talking about a website, but about a nodejs server.
Nos vemos por la red.

logix5, 2020-10-22:
Thanks Antonio,

I have the blocks in my server
<https://logix5.com/Blockyduino-para-MRTDuino/demos/code/index.html>, and
it is very interesting to do more things. I will read your recommendations.

I would like to add more blocks to otto

Br

-- 
Óscar Ferruz Romero
*Socio fundador.*
*Tel. 620.402.168*

 C/ Artajona, 17.  28039 - Madrid.
www.logix5.com
blog.logix5.com

www.twitter.com/Logix5_SS <http://www.twitter.com/logix5_SS>

www.facebook.com/logix5

Instagram: logix5_

"En la vida hay tres cosas que no dan marcha atrás: la palabra pronunciada,
la flecha lanzada y la oportunidad perdida, ¿vamos a dejarla escapar?"

cparrapa, 2020-10-23:
cool @agomezgar  so your now works online? can upload code from any OS?

agomezgar, 2020-10-26:
I'm afraid I've not been clear enough. I meant that Electron is a technology which allows you to convert a website to an app functional in any OS. Sorry for the misunderstanding.

cparrapa, 2020-10-26:
no worries @agomezgar the only OS we have not figure out is Mac for some reason is not that simple to just tell electron to pack to certain OS specially because the arduino builder or CLI needs to be different for each OS

cparrapa, 2020-12-08:
solved here https://github.com/OttoDIY/blockly/pull/66

closed 2020-12-08 by cparrapa