OttoDIY/blockly #61: Error función getdistance

NeftaliVenancio, 2020-10-17:
Realizando pruebas en la versión online de blocky, me encontré que la función por defecto del sensor ultrasónico me devuelve valores como si no encontrará la función específica, caso contrario generando el objeto y especificandole en que pines de encuentra el sensor de forma manual la lectura es totalmente correcta, he estado revisando las librerías de Otto que me arrojan dependencia pero no logro encontrar donde se ejecuta el cálculo exactamente.

cparrapa, 2020-10-20:
Hola 
no te aparece la distancia?

NeftaliVenancio, 2020-10-20:
No, aparece un único valor de 999.0, como si tuviera un valor constate en
una variable y nunca realizará la lectura para actualizarlo, algo asi me
paso al mandar a llamar el Acelerómetro, pero cuando mandaba a llamar por
sí solo al acelerómetro se generaba la lectura y con eso se soluciona, pero
en el ultrasonico no encuentro una función similar. descargue todas las
librerías necesarias para compilar los códigos de blocky en VS code, pero
no logro dar con la función original que calcula la distancia.

cparrapa, 2020-10-20:
A mi eso solo me pasaba cuando el ultrasonido estaba malo o conectado donde no era. Donde lo tienes conectado?

NeftaliVenancio, 2020-10-20:
Lo mismo pense, está conectado como lo marca el código de blocky

#define PIN_Trigger 8 // ultrasound
#define PIN_Echo 9 // ultrasound

Como pense lo mismo, que el sensor estuviera dañado, hice la prueba
manualmente respetando las conexiones, únicamente usando una librería
externa y funciono correctamente, de igual forma hice la prueba con el
constructor de objeto en blocky, te adjunto una captura

[image: Captura de Pantalla 2020-10-20 a la(s) 16.11.30.png]

Haciendo esto de construir el objeto, al revisar el código genera una
función específica y ahí si funciona correctamente, en los tres casos hice
la prueba con varios sensores, en la única en que no funcionaba
correctamente es usando el bloque dedicado para la distancia. donde manda a
llamar la función getdistance(), te deje otra captura:

[image: Captura de Pantalla 2020-10-20 a la(s) 16.14.27.png]

Solo lo he podido probar en la versión online usando safari, no he podido
probar en escritorio puesto que estoy en macOS.

cparrapa, 2020-10-26:
Weird, no puedo ver tus fotos puedes indicar en que parte del codigo ves el error? o puedes hacer un PR?

gracias

cparrapa, 2021-01-06:
Hola @NeftaliVenancio 
Ya debe estar solucionado en la nueva version https://github.com/OttoDIY/blockly/releases/tag/v1.4.0

closed 2021-01-06 by cparrapa