OttoDIY/blockly #62: Some bugs found in version 1.3.0

adalborgo, 2020-10-30:
I installed the new version of Otto Blockly 1.3.0 (OttoBlockly-Setup-1.3.0.exe) but I encountered some bugs:
- When I click on **New**, the workspace is only partially cleared.
- When you change the device on the USB port, the port number does not change.
- When you drag the **Loop** block, some elements detach, so you can neither delete nor reconnect them to the original block.

In all cases it is necessary to leave Otto Blockly and re-enter.

cparrapa, 2020-10-30:
Strange none of those happened to me, what windows are you using?

adalborgo, 2020-11-01:
My operating system is Windows 10 Pro (Build: 19041.572)

To check the conditions of the problems indicated I have prepared some files that you can find at the following addresses:
https://dibis.it/download/tmp/ButtonErr.zip
https://dibis.it/download/tmp/PulsanteOk.bloc
https://dibis.it/download/tmp/ReloadErr.zip

A. ButtonErr.zip contains the script which gives the indicated problem and 4 screenshots:
OttoBlockly-a1.png: the blocks with the Arduino script
OttoBlockly-a2.png: after being saved and reloaded, a Loop block is dragged
OttoBlockly-a3.png: I click on New
OttoBlockly-a4.png: what remains in the workspace after New.

B. The ButtonOk.bloc script, on the other hand, does not give the problems of the previous one, but if I overwrite it and reload it I get an empty workspace;
however, if I exit and reload it, everything is ok again.

cparrapa, 2021-01-04:
Hi @adalborgo 
Update your blockly we made some changes that could have solved this

cparrapa, 2021-01-06:
new version solved this @adalborgo
https://github.com/OttoDIY/blockly/releases/tag/v1.4.0

closed 2021-01-06 by cparrapa