OttoDIY/blockly #63: Italian translation

adalborgo, 2020-10-30:
I would like to know whether the **Italian** translation is in progress, otherwise I could start working on it.

cparrapa, 2020-10-30:
Ciao @adalborgo 

Not in progress 
grazie!

adalborgo, 2020-11-21:
I have completed the **Italian translation**, but before making it final I prefer to check it in my context.
How can I insert it into my local installation?

cparrapa, 2020-11-21:
Ciao @adalborgo 
You have to go to your instalattion folder and modify this file https://github.com/OttoDIY/blockly/blob/master/www/lang/code.js

add Italian and make sure your new files are also in the same folder with the right name like this
Arduino_it.js Blockly_it.js and msg_it.js files

Grazie looking forward for your translation

cparrapa, 2021-01-04:
Hi @adalborgo

Happy new year let me know if you still want to help us with the Italian translation we are gonna release the new 1.4 version so if you want to have Italian on it we need to do it this week.

Thanks

adalborgo, 2021-01-05:
You can download the translation from this address:
https://www.dibis.it/download/OttoDIY/Italian.zip

adalborgo, 2021-01-06:
Sorry, I found an error in the translation of the **Arduino_it.js** file, please reload the file 'Italian.zip' from https://www.dibis.it/download/OttoDIY/Italian.zip

Thanks.

adalborgo, 2021-01-11:
Last week I made the translation of the OttoBlockly messages available, but the same day the new version came out with the modified Arduino_xx.js file.
I have already started preparing the new translation, but could you please send me the new files in advance with the next version.
Thanks.

adalborgo, 2021-01-17:
The Italian translation of version 1.4 is available at this address:
https://www.dibis.it/download/OttoDIY/Italian-1.4.zip