OttoDIY/blockly #72: Bug in OLED block

rsparkie, 2020-12-19:
I'm just checking out the latest version of Otto Blockly 'Otto Nuevo' and I was having difficulty updating text in the OLED.  I used the 'clear OLED' block to wipe the screen before applying the new text, but I'm finding this wasn't working.

To send a command to make some text on the OLED such as 'hi there' there are 2 lines of Arduino code;
display.println("hi there");
  display.display();

The arduino code currently called by the 'clear OLED' block is: display.clearDisplay();
but it doesn't work. So when I add a second line 'display.display();' it now works

So the code to clear the display is;

display.clearDisplay();
  display.display();

and this works

In addition there is an initialization block for the OLED, and the corresponding block has the code;

display.begin(SSD1306_SWITCHCAPVCC, 0x3C);
display.clearDisplay();

I understand this should also be modified to add 'display.display();' so it becomes;
;
display.begin(SSD1306_SWITCHCAPVCC, 0x3C);
display.clearDisplay();  
display.display();

I am using arduino Leonardo but I noticed the block is the same on other boards also

logix5, 2020-12-19:
Hi !!

These oled blocks were not updated. I have not test it. I have added other
blocks or features, but the OLED blocks were nor updated.

When i use a OLED block i will test too

Thanks

-- 
Óscar Ferruz Romero
*Socio fundador.*
*Tel. 620.402.168*

 C/ Artajona, 17.  28039 - Madrid.
www.logix5.com
blog.logix5.com

www.twitter.com/Logix5_SS <http://www.twitter.com/logix5_SS>

www.facebook.com/logix5

Instagram: logix5_

"En la vida hay tres cosas que no dan marcha atrás: la palabra pronunciada,
la flecha lanzada y la oportunidad perdida, ¿vamos a dejarla escapar?"

logix5, 2020-12-22:
Hi

I have reviewed the documentation , and you are right. this line must be
added

Information about clearDisplay

/*!
@brief Clear contents of display buffer (set all pixels to off).
@return None (void).
@note Changes buffer contents only, no immediate effect on display.
Follow up with a call to display(), or with other graphics
commands as needed by one's own application.
*/ I am going to update this block Thanks !!

-- 
Óscar Ferruz Romero
*Socio fundador.*
*Tel. 620.402.168*

 C/ Artajona, 17.  28039 - Madrid.
www.logix5.com
blog.logix5.com

www.twitter.com/Logix5_SS <http://www.twitter.com/logix5_SS>

www.facebook.com/logix5

Instagram: logix5_

"En la vida hay tres cosas que no dan marcha atrás: la palabra pronunciada,
la flecha lanzada y la oportunidad perdida, ¿vamos a dejarla escapar?"

cparrapa, 2020-12-23:
fixed here https://github.com/OttoDIY/blockly/commit/f73b1d29ff1581c58ee65ba1cd37436469a43b57

closed 2020-12-23 by cparrapa