OttoDIY/blockly #80: Block for OLED animations

cparrapa, 2021-01-02:
We have already a block for OLED bitmaps how to make it animations like this video? 
https://youtu.be/0KGMFhFQ0YY

cparrapa, 2021-02-26:
@cadcygnus It is true that the bmp for the oled needs to rewritten in code as bmp1, bmp2, etc, for the various images to show up on the oled. Then would it not be more feasible if you created a block that will automatically rename the bmp to the ordered sequence, like you did in the function blocks? Then instead of having you write a program for everyone who needs to use more than one image in the oled you can have the block write in the code. 

I think this would improve your blockly to do animations and even gifs.

logix5, 2021-02-27:
https://www.instructables.com/Animation-on-OLED-Arduino/

-- 
Óscar Ferruz Romero
*Socio fundador.*
*Tel. 620.402.168*

 C/ Artajona, 17.  28039 - Madrid.
www.logix5.com
blog.logix5.com

www.twitter.com/Logix5_SS <http://www.twitter.com/logix5_SS>

www.facebook.com/logix5

Instagram: logix5_

"En la vida hay tres cosas que no dan marcha atrás: la palabra pronunciada,
la flecha lanzada y la oportunidad perdida, ¿vamos a dejarla escapar?"

cadcygnus, 2021-02-28:
Can you convert this into a blockly?  I saw your video and program. Nice work.

logix5, 2021-02-28:
Hi

Maybe we can add blocks as ST7735, where you can create variables for
frames and then you can draw.  these blocks:

[image: image.png]

We can clone these blocks in OLED display

-- 
Óscar Ferruz Romero
*Socio fundador.*
*Tel. 620.402.168*

 C/ Artajona, 17.  28039 - Madrid.
www.logix5.com
blog.logix5.com

www.twitter.com/Logix5_SS <http://www.twitter.com/logix5_SS>

www.facebook.com/logix5

Instagram: logix5_

"En la vida hay tres cosas que no dan marcha atrás: la palabra pronunciada,
la flecha lanzada y la oportunidad perdida, ¿vamos a dejarla escapar?"

cparrapa, 2021-03-09:
It looks to me great progress in this commit https://github.com/OttoDIY/blockly/commit/da3ed5928045c1af6a6568fd857495277c36f093

In the video a guy uses this software to create even more smooths animations can this be just copied inside the block?
https://www.youtube.com/redirect?event=video_description&redir_token=QUFFLUhqbnYwM3NQdEFxRkR2OFRKS1FISWpVeDJrVHVhQXxBQ3Jtc0tuMW9LTXFHVXI3TVRKTk1YSVV5TGU2aVc3eUZ5QldHU2NGSi04ZV9ra1pkSWp0dW5jYVdLbklSbnFESzlhVTJKaHFiNUNjMmVhbXBnMWtvbU9ob3RtSWlBTk8wUkRKTlg5elZRT0k5Nmt2aFlkMVRkRQ&q=https%3A%2F%2Fdrive.google.com%2Fopen%3Fid%3D1ECufN2p8tpTUwq2UFgZK_Rq3wDmOACDo

So far i use this tool but is only for frame per frame not the entire animation https://javl.github.io/image2cpp/

 i found this cool tool for creating OLED animations, basically you can just pick web icons8 that already have the frames and here you can convert them to code https://wokwi-app.web.app/animator

closed 2021-06-13 by cparrapa