OttoDIY/blockly #91: Suggestion to support workshops - capability checklist

brendonhatcher, 2021-01-10:
In setup, there is currently options for user experience and available blocks.

I think it would be great if there was a section where you list your Otto's capabilities, and that either hides the blocks you can't use, or marks them as unsuitable.  Capabilities would include all the variants and common optional hardware - arms, neopixel, distance sensor etc.

As an added advantage, workshop leaders could encourage participants to fill this list in themselves, leading to a more structured exploration of the robot by the user.

cparrapa, 2021-01-10:
Hi 
Do you mean like have a setup per kit like in the ottodiy.com website?
[image]

It is more or less what we tried with the setup menu with the check list, not sure if you mean a different layout? there are many additional blocks that we are exploring to incorporate into Otto curriculum one by one but they are already useful for exploration for the builders to combine or for any Arduino based project unrelated to Otto.

cparrapa, 2021-06-06:
Hi @brendonhatcher 

We are making Otto modular so is more of what blocks you add=expansions to put
For example:
Otto Starter+ Led mouth and arms expansion = Otto Humanoid, then you just "add" the extra blocs you want to use

What do you think of this new approach and layout? https://ottodiy.github.io/blockly/www/

closed 2021-06-13 by cparrapa