OttoDIY/blockly #92: New devices

adalborgo, 2021-01-11:
I am adding some devices to OttoBlockly, in particular for the "[Seeed Studio Grove Beginner Kit for Arduino](https://files.seeedstudio.com/wiki/Grove-Beginner-Kit-For-Arduino/img/Parts.jpg)", when I'm done I will send the code.

cparrapa, 2021-03-14:
great looking forward @adalborgo

cparrapa, 2021-05-01:
> I am adding some devices to OttoBlockly, in particular for the "[Seeed Studio Grove Beginner Kit for Arduino](https://files.seeedstudio.com/wiki/Grove-Beginner-Kit-For-Arduino/img/Parts.jpg)", when I'm done I will send the code.

Hi @adalborgo 
Thanks where can we find those new blocks?

adalborgo, 2021-05-08:
I have prepared a zipped file with all the necessary files, you can download it from this link:
https://www.dibis.it/download/OttoDIY/GroveBegKit.zip

logix5, 2021-05-08:
Hi Antonio,

I have a question, what board do you select in the board menu to upload the
code ?

you are using this board https://wiki.seeedstudio.com/Seeeduino_Lotus/
right?

BR

-- 
Óscar Ferruz Romero
*Socio fundador.*
*Tel. 620.402.168*

 C/ Artajona, 17.  28039 - Madrid.
www.logix5.com

www.twitter.com/Logix5_SS <http://www.twitter.com/logix5_SS>

www.facebook.com/logix5

www.instagram.com/logix5_smart_solutions

"En la vida hay tres cosas que no dan marcha atrás: la palabra pronunciada,
la flecha lanzada y la oportunidad perdida, ¿vamos a dejarla escapar?"

logix5, 2021-05-08:
Hi Antonio again

Your board is arduino uno compatible, i have readed this information. I
think you can use the current sensor/actuators because the most of your
blocks are using led, potentiometer, light sensor,... You only need update
the name of the pins.

We can add the SSD35 and lis3dhtr blocks because these sensors or libraries
are not present in the blockly. The rest sensors you can use them.

I can create a simple block to convert D1, Dx in the pin of the board.  But
we are trying to create a customize configuration depending of the type of
board.

Let us know

-- 
Óscar Ferruz Romero
*Socio fundador.*
*Tel. 620.402.168*

 C/ Artajona, 17.  28039 - Madrid.
www.logix5.com

www.twitter.com/Logix5_SS <http://www.twitter.com/logix5_SS>

www.facebook.com/logix5

www.instagram.com/logix5_smart_solutions

"En la vida hay tres cosas que no dan marcha atrás: la palabra pronunciada,
la flecha lanzada y la oportunidad perdida, ¿vamos a dejarla escapar?"

adalborgo, 2021-05-08:
Hi Oscar,
Seeeduino Lotus is compatible with Arduino UNO, but also has analog pins A6 and A7 not available on the connector of UNO.
Rather than creating a custom configuration for that type of board, it is easier to assign the available pins in the 'Arduino_xx.js' (see 'grove_beginner-msg_xx.js') file. This way you can customize the order and put the default one at the beginning.
However, the pins are fixed in the Grove beginner's kit.

cparrapa, 2021-06-06:
> I have prepared a zipped file with all the necessary files, you can download it from this link:
> https://www.dibis.it/download/OttoDIY/GroveBegKit.zip

Thanks @adalborgo can you do a PR please?
Like this is easier for us to merge

closed 2021-07-15 by cparrapa