OttoDIY/blockly #93: Release type

adalborgo, 2021-01-11:
OttoBlockly doesn't need a proper installation, so I suggest releasing a zipped version as well.
Normally I unzip the .exe file with 7zip and copy only the files contained in app-32.7z.

cparrapa, 2021-01-12:
Hi
We use to have it but people got confused and started downloading both, you can download the whole repo that is basically the same thing

adalborgo, 2021-01-17:
With the addition of new devices, version 1.4 has become too big (over 2GB), I think it would be better to have some lighter versions as well. I have seen that it is quite easy to make changes without having to recompile the whole package.

logix5, 2021-01-17:
Yes you are right, but the size is due to add arduino-cli and several
boards. The compilation folder is the reason of this increase, but it
doesn't affect the performance.

Maybe we can create a tool to delete the boards or libs that you don´t want
use. But in my opinion there is no problem because only last in the
installation process.

Now i have added the GPS sensor and soon we will add more features

BR

-- 
Óscar Ferruz Romero
*Socio fundador.*
*Tel. 620.402.168*

 C/ Artajona, 17.  28039 - Madrid.
www.logix5.com
blog.logix5.com

www.twitter.com/Logix5_SS <http://www.twitter.com/logix5_SS>

www.facebook.com/logix5

Instagram: logix5_

"En la vida hay tres cosas que no dan marcha atrás: la palabra pronunciada,
la flecha lanzada y la oportunidad perdida, ¿vamos a dejarla escapar?"

adalborgo, 2021-05-13:
Seeeduino Lotus is perfectly compatible with Arduino UNO.

cparrapa, 2021-06-06:
Alright for next release we will add the .zips thanks! @adalborgo

closed 2021-06-06 by cparrapa