OttoDIY/blockly #95: Build OttoBlockly for Windows

adalborgo, 2021-01-27:
I tried to build OttoBlockly for Windows from source but without the expected result.
Especially the buttons maximize, minimize, restore and USB port selection doesn't work.
The installation build processes report no errors.
Any suggestions? Thanks.

logix5, 2021-01-28:
Hi Antonio

Can you review the index.html file in www folder. Maybe in the installation
process the web file was copied.  Can you share with us a screenshot
because if the USB port have disappeared the index file used is not ok

Let us know

-- 
Óscar Ferruz Romero
*Socio fundador.*
*Tel. 620.402.168*

 C/ Artajona, 17.  28039 - Madrid.
www.logix5.com
blog.logix5.com

www.twitter.com/Logix5_SS <http://www.twitter.com/logix5_SS>

www.facebook.com/logix5

Instagram: logix5_

"En la vida hay tres cosas que no dan marcha atrás: la palabra pronunciada,
la flecha lanzada y la oportunidad perdida, ¿vamos a dejarla escapar?"

adalborgo, 2021-01-28:
I think it is a version problem.
In order not to have errors with serialport, I compiled it with the "--build-from-source" option, but probably the new version 9.0.6 is not compatible with the OttoBlockly code.
I also edited the package.json file in the script section, from "compiler": "build --win --ia32" to "compiler": "electron-builder --win --ia32", because npm does not recognize the word "build".
It would be necessary to review the versions of the packages because those indicated are not up to date.

I also did some tests with Blocklino but the problem is the same.

adalborgo, 2021-02-01:
The problem with the maximize, minimize, reset buttons only occurs with Electron versions greater than 4: some statements in the code are obsolete.

As for the SerialPort, version 9.0.6 does not see serial ports.
In the documentation of "https://serialport.io/docs/api-stream" I found these notes:
- In serialport@8 and @serialport/bindings@3 **we renamed PortInfo.comName to PortInfo.path**. If you use comName you'll get a warning until the next major release.
 - In serialport@8 and @serialport/bindings@3 **we removed the optional callback to list()**. You now have to use the promise it returns and if you provide a callback you will get an error.

In the **resources/app/index.js** file:

_lines of code 28-52_:
	document.getElementById('#portserie').mouseover(function(){ // $('#portserie').mouseover(function(){
		sp.list(function(err,ports) {
			var nb_com = localStorage.getItem("nb_com"), menu_opt = portserie.getElementsByTagName('option')
			if(ports.length > nb_com){
				ports.forEach(function(port){
					if (port.vendorId){
						var opt = document.createElement('option')
						opt.value = **port.comName**
						opt.text = **port.comName**
						portserie.appendChild(opt)
						localStorage.setItem("com",port.comName)
					}
				})
				localStorage.setItem("nb_com",ports.length)
				localStorage.setItem("com",portserie.options[1].value)
			}
			if(ports.length < nb_com){
				while(menu_opt[1]) {
					portserie.removeChild(menu_opt[1])
				}
				localStorage.setItem("com","com")
				localStorage.setItem("nb_com",ports.length)
			}
		})
	})

_lines of code 94-114_:
	sp.list(function(err,ports){
		var opt = document.createElement('option')
		opt.value = "com"
		opt.text = Blockly.Msg.com1
		portserie.appendChild(opt)
		ports.forEach(function(port) {
			if (port.vendorId){
				var opt = document.createElement('option')
				opt.value = **port.comName**
				opt.text = **port.comName**
				portserie.appendChild(opt)
			}
		})
		localStorage.setItem("nb_com",ports.length)
		if (portserie.options.length > 1) {
			portserie.selectedIndex = 1
			localStorage.setItem("com",portserie.options[1].value)
		} else {
			localStorage.setItem("com","com")
		}
	})
	
Is it possible to modify the index.js file to make it compatible with the new version of SerialPort?

adalborgo, 2021-02-05:
I have found the solution. It is necessary to replace the lines I indicated with:
_lines of code 28-52_:

	$('#portserie').mouseover(function(){
			sp.list().then(ports => {
			var nb_com = localStorage.getItem("nb_com"), menu_opt = portserie.getElementsByTagName('option')
			if(ports.length > nb_com){
				ports.forEach(function(port){
					if (port){
						var opt = document.createElement('option')
						opt.value = port.path
						opt.text = port.path
						portserie.appendChild(opt)
						localStorage.setItem("com",port.path)
					}
				})
				localStorage.setItem("nb_com",ports.length)
				localStorage.setItem("com",portserie.options[1].value)
			}
			
			if(ports.length < nb_com){
				while(menu_opt[1]) {
					portserie.removeChild(menu_opt[1])
				}
				localStorage.setItem("com","com")
				localStorage.setItem("nb_com",ports.length)
			}
		})
	})

_lines of code 94-114_:

	sp.list().then(ports => {
		var opt = document.createElement('option')
		opt.value = "com"
		opt.text = Blockly.Msg.com1
		portserie.appendChild(opt)
        ports.forEach(function (port) {
			if (port){
            var opt = document.createElement('option')
            opt.value = port.path;
            opt.text = port.path;
            portserie.appendChild(opt)
			}
        });
		
		localStorage.setItem("nb_com",ports.length)
		if (portserie.options.length > 1) {
			portserie.selectedIndex = 1
			localStorage.setItem("com",portserie.options[1].value)
		} else {
			localStorage.setItem("com","com")
		}
	})

cparrapa, 2021-03-14:
Hi @adalborgo 
Looking good thanks for the debugging can you do a PR with this changes so that we can try?

cparrapa, 2021-05-01:
> I have found the solution. It is necessary to replace the lines I indicated with:
> _lines of code 28-52_:
>
> ```
> $('#portserie').mouseover(function(){
> 		sp.list().then(ports => {
> 		var nb_com = [...]

Hi @adalborgo what files is this?

cparrapa, 2021-06-13:
Hi @adalborgo 
Give a try to this new release https://github.com/OttoDIY/blockly/releases/tag/v1.4.2

quangvtvp, 2021-07-12:
@cparrapa I would like to build for window with my small of custom otto lib, so could you share the detail of instructions to build for window os?

cparrapa, 2021-07-12:
Hi @quangvtvp 
The instructions are in the readme, if complicated we can add it for you. is it a new Otto remix?

quangvtvp, 2021-07-14:
Hi @cparrapa, I have already checked the readme file but it not found any step instruction to build exe file for window, could u give me a link?

closed 2021-07-15 by cparrapa