OttoDIY/blockly #96: Logic and Math menu bug in the Portuguese version

lsprandel, 2021-02-13:
Hello,
when we are using on portuguese translation the menu of Logic and Math doesnt show correctly. This failure compromises the use in this language. 
Thanks a lot. 

[image]
[image]

cparrapa, 2021-02-14:
Thanks for letting us know @lsprandel

fixed here https://github.com/OttoDIY/blockly/commit/4fab97787b20566399432b4475bf130c986537ac

Download this file https://github.com/OttoDIY/blockly/blob/master/www/lang/Blockly_pt.js and replace in your installed blockly 

Let me know if works for you

closed 2021-02-14 by cparrapa

elton-bt, 2021-11-05:
The solution works!

commit 4fab9778 (2021-02-14): fix #96 and clean examples