OttoDIY/blockly #97: Blocks for multiple stepper motors

cparrapa, 2021-03-02:
Right now we can only control one stepper motor, we need at least 2 to have a maze solver robot with wheels. Something like this
[image]

logix5, 2021-03-02:
I will migrate my stepper block, because i can use several.

I let you know

-- 
Óscar Ferruz Romero
*Socio fundador.*
*Tel. 620.402.168*

 C/ Artajona, 17.  28039 - Madrid.
www.logix5.com

www.twitter.com/Logix5_SS <http://www.twitter.com/logix5_SS>

www.facebook.com/logix5

www.instagram.com/logix5_smart_solutions

"En la vida hay tres cosas que no dan marcha atrás: la palabra pronunciada,
la flecha lanzada y la oportunidad perdida, ¿vamos a dejarla escapar?"

logix5, 2021-03-03:
Multiple stepper motors were added. No tested yet, but the compilation was
ok

any problem please let me know

-- 
Óscar Ferruz Romero
*Socio fundador.*
*Tel. 620.402.168*

 C/ Artajona, 17.  28039 - Madrid.
www.logix5.com

www.twitter.com/Logix5_SS <http://www.twitter.com/logix5_SS>

www.facebook.com/logix5

www.instagram.com/logix5_smart_solutions

"En la vida hay tres cosas que no dan marcha atrás: la palabra pronunciada,
la flecha lanzada y la oportunidad perdida, ¿vamos a dejarla escapar?"

closed 2021-03-14 by cparrapa