OttoDIY/blockly #98: PopUp Status UX

gersonsena, 2021-04-29:
Helo,

I'm from Brazil, and I founded a great community about OttoDIY (using Otto Blockly). And I would like to leave my humble suggestion for improvement.

The status popup needs press on the little "x" to closed it. Without this, the screen interface does not works (hangs), and it needs closed the popup every time. This is a bit annoying, when you need to write the same code over and over and have to close all those times by moving the mouse. 

Would there be a way to change this popup to a different type of status window? It doesn't really help at all in the user experiences.

I'm sorry if I seem too layman with this or I'm upsetting you, it's just my opinion of using it as a children's teacher.

I'm sorry even if I'm being boring. You guys do a great job, and congratulations for that.

gersonsena, 2021-04-29:
In time, this is the popup status I was talking about (above)

[image]

cparrapa, 2021-05-01:
Hi @gersonsena 

You are totally right it is really annoying that X and window, the bad news is i don't know how to fix. :(

Do you know how can we change this?
Thanks for reporting

logix5, 2021-05-01:
I will try to investigate the source code :-)

I let you know

-- 
Óscar Ferruz Romero
*Socio fundador.*
*Tel. 620.402.168*

 C/ Artajona, 17.  28039 - Madrid.
www.logix5.com

www.twitter.com/Logix5_SS <http://www.twitter.com/logix5_SS>

www.facebook.com/logix5

www.instagram.com/logix5_smart_solutions

"En la vida hay tres cosas que no dan marcha atrás: la palabra pronunciada,
la flecha lanzada y la oportunidad perdida, ¿vamos a dejarla escapar?"

gersonsena, 2021-05-02:
Hi, @cparrapa 
Unfortunately, I wouldn't even know how to start fixing it.

@logix5 
I really appreciate your help.  Thanks. I wish success at work

cparrapa, 2021-06-06:
Hi Guys @gersonsena @logix5 
I managed to put a 3 seconds timer for the windows to go off, let me me know if that will do!

closed 2021-06-06 by cparrapa

gersonsena, 2021-06-07:
Hi @cparrapa 
I will redownload OttoBlockly when I am ready to change and test. But I think this helps.

cparrapa, 2021-06-13:
@gersonsena how did it go the test? here you can download the latest https://github.com/OttoDIY/blockly/releases/tag/v1.4.2

gersonsena, 2021-06-15:
Hi, Camilo!

It worked here. I downaloded the version from link that you send.

It's better than before. Congratulations.

Hug

-- 
*Gerson Sena*
Engenheiro Eletricista - CREA-RS 236433
Especialista em Metodologias de Ensino da Matemática e da Física
CV Lattes: ​http://lattes.cnpq.br/2879064516987039
<http://lattes.cnpq.br/2879064516987039>

​CEO da Aplicar Ciências e Robótica Educacional
*​Site:* https://www.aplicarciencias.com​
*Canal:* https://www.youtube.com/c/aplicarcienciaserobotica
*Telegram (Canal):* https://t.me/aplicarciencias
*Instagram:* https://www.instagram.com/aplicar_ciencias_robotica
*Facebook*: https://www.facebook.com/aplicarciencias/
*WhatsApp:* *+55 55 999 191 349*