OttoDIY/blockly #100: error when checking the code and upload with the language in Portuguese

elton-bt, 2021-05-04:
Using the Linux version on Ubuntu 21.04, everything works when it is in English. However, when I change the interface to Portuguese, and I click on check the code, it appears the following error message:

ERROR DE COMPILACIÓN/home/alice/.OttoBlockly/arduino/sketch/sketch.ino: In function 'void loop()': /home/alice/.OttoBlockly/arduino/sketch/sketch.ino:17:29: expected primary-expression before ';' for (int count=0 ; count< ; count++) { ^ Error during build:
nano

Even if the error occurs, when you click the upload button for the first time, the code is sent.
When you click the button a second time, even without changing the code, the erro message appears: blocks not connected.

I'm available for testing. Thank you and congratulations for this greate project.

cparrapa, 2021-05-10:
Hi @elton-bt try updating the language and blocks for the linux version with the windows one

elton-bt, 2021-05-11:
Hi @cparrapa , thanks for the reply. 

To update I copied the blocs & generateurs and lang folders from the Windows version to the folder with the my Linux version. Was It that way to update?

Anyway, the error persists.

I also noticed that both folders in the Linux version have fewer files than those in the Windows version.

cparrapa, 2021-05-12:
Hi @elton-bt  
Can you post a screenshot of your code? (both blocks and text code)

Try grabbing all the folders from the Windows version here https://github.com/OttoDIY/blockly/tree/master/www
if not it must be an issue in the Portuguese files you can check in the lang folder what are the differences

elton-bt, 2021-05-12:
Hi again, @cparrapa . Here are the [screenshot.](https://pasteboard.co/K1ygDn3.png) I am using the first example code present in the program itself, the walk program.

I replaced all the files in my www folder with the Windows version, but the problem still persists. When I change the interface to English, everything works.

cparrapa, 2021-05-12:
it looks to me you are using old blocks that we don't use anymore, you have not updated the toolbox folder

elton-bt, 2021-05-12:
Hello @cparrapa , I believe I found the problem (but not how to solve it).

I downloaded the project again and followed all the steps for installing on Linux. I did the tests with English and everything was ok. I deleted the www folder from my installation and replaced it with the www folder of the Windows version, and even then the problem happens.

However, on closer inspection, the problem only occurs with logic blocks. [Here ](https://pasteboard.co/K1yUTS2.png)is the code without logic blocks in portuguese. [Here](https://pasteboard.co/K1yVt3X.png), you can see on the left, that they are together and I can't drag.

In addition, there is a difference between these blocks and [English](https://pasteboard.co/K1yVLeY.png) and [Portuguese](https://pasteboard.co/K1yVUE3.png) for the same sample code. In Portuguese, logic blocks do not show the number of times of repetition. Math blocks are also having the same problem in Portuguese. All other blocks are ok in Portuguese.

elton-bt, 2021-05-12:
I have a windows pc to test, but I was unable to install blockly, I think the installation instruction was missing as it exists for linux.

cparrapa, 2021-05-12:
I don't have a Linux to test 🤒  
keep checking and trying I believe the Linux version is outdated with the main branch to install Windows you only need to download the latest release .exe

elton-bt, 2021-05-13:
Ok, and where can I find the release.exe? I downloaded the master brench and didn't find this file.

As a suggestion, i would find it important to clarify how to install on Windows.

cparrapa, 2021-05-16:
in the release section https://github.com/OttoDIY/blockly/releases is just an .exe the software will guide you in every step of the installation

elton-bt, 2021-05-17:
Thank you for clarifying about the installation. I was able to install and run normally on Windows with the Portuguese language. I think the problem is only with the Linux + Portuguese version.

cparrapa, 2021-06-06:
Does this happen to you guys?
@agomezgar @gitaroktato @mxochicale

mxochicale, 2021-09-04:
Sorry for the late reply. @elton-bt  have you tried to run the English version on your machine with your otto? I guess having done that test will help to then follow up any issues in the translated app in Portuguese. Hope that helps!

elton-bt, 2021-09-10:
Hi @mxochicale. As described in the previous post, the English version works perfectly. The problem is only in the Portuguese version, because, apparently, the logical and mathematical blocks are in trouble.

> Hello @cparrapa , I believe I found the problem (but not how to solve it).
>
> I downloaded the project again and followed all the steps for installing on Linux. I did the tests with English and [...]

mxochicale, 2021-09-10:
Thanks @elton-bt for the clarifications.  Have you tried to compare the code for the English version against your Portuguese version?. Maybe you are using some special characters in those logic functions? maybe you need a special library?. Not sure how @cparrapa is managing the multiple languages of the app (or if there is any guidance on how to do that) but it would be nice that you also add your Portuguese translation for the GNU/Linux version of the app.

elton-bt, 2021-09-10:
Hi, @mxochicale. [Here](https://pasteboard.co/K1yUTS2.png) you can see the code running in Portuguese without problems. The error only happens when using logical or mathematical blocks.
In [this](https://pasteboard.co/K1yVt3X.png) image, you can see that the logic blocks are overlapping each other, not allowing them to be dragged. If I create a code in the English interface using these blocks, when switching to Portuguese, the code has an error when compiling.
Unfortunately I don't have the necessary skills to try to fix the code that causes these logical and mathematical blocks to be "glued" in the Portuguese interface. I can help with the translation and proofreading for Portuguese.
Thanks

cparrapa, 2021-10-30:
Hi @mxochicale @elton-bt @Pawka solved here https://github.com/OttoDIY/blocklyLinux
thanks to @agomezgar  for this great contribution

closed 2021-10-30 by cparrapa

mxochicale, 2021-10-30:
:fire: Many thanks @cparrapa and @agomezgar for such update.
I will have a go to the new repo sometime in Nov2021!

PS. you might like to delete the branch of https://github.com/OttoDIY/blockly/tree/versionlinux

elton-bt, 2021-11-04:
Hi, @mxochicale @cparrapa @Pawka unfortunately the error remains as can be seen in the [image](https://pasteboard.co/bZ7CPI60tTI7.png). I downloaded the project from the linux branch again, and the same logical and mathematical blocks are still "glued" when the application language is in Portuguese.

elton-bt, 2021-11-04:
The same error was related here: https://github.com/OttoDIY/blockly/issues/96 in February.

cparrapa, 2021-11-04:
> 🔥 Many thanks @cparrapa and @agomezgar for such update. I will have a go to the new repo sometime in Nov2021!
> 
> PS. you might like to delete the branch of https://github.com/OttoDIY/blockly/tree/versionlinux

Hi @mxochicale thanks mainly to @agomezgar he did it all! 
How to delete the branch without deleting the entire repo? i did not made that branch so not sure how to delete

cparrapa, 2021-11-04:
> The same error was related here: #96 in February.

Hi @elton-bt you can see the solution is right there too
 https://github.com/OttoDIY/blockly/blob/master/www/lang/Blockly_pt.js

elton-bt, 2021-11-05:
Thanks @cparrapa . The solution works.