OttoDIY/blockly #101: Error when upload sketch via usb by using Otto blockly web

quangvtvp, 2021-05-13:
I use Ardruino nano old bootloader. when I upload sketch via usb get error:

Command failed: /home/vcap/app/bin/arduino-cli compile --fqbn arduino:avr:nano:cpu=atmega328old --output-dir uploads/arduino1620896806726/build uploads/arduino1620896806726 Error during build: platform not installed

Please help me find the solution to solved this issue.

cparrapa, 2021-05-16:
try using the simple arduino nano, if problem persist show us your code

quangvtvp, 2021-05-18:
@cparrapa just changed to arduino nano but the problem still there.
This is the simple code I use and error image. please help me check it. Thanks!
[Image 1](https://drive.google.com/file/d/1Gr4GSbZkIUzXN7ASCEDhryjgWuvmJH_O/view?usp=sharing)
[Image 2](https://drive.google.com/file/d/1-YsL5yN9KMXJKKXbcuu__7N4nqbJWFD9/view?usp=sharing)

cparrapa, 2021-05-19:
Hi @quangvtvp 
I see there is a mistake in the code rather than the software you are missing the "end" block
Try also this new link and let me know https://ottoschool.com/blockly2/

quangvtvp, 2021-05-19:
@cparrapa thanks for feedback.
I tried to used : https://ottoschool.com/blockly2/ but still have an error.
[Image](https://drive.google.com/file/d/1DzFM0deUganXBp1puNGC71_MFACrg6nv/view?usp=sharing)
I opened the example "Walk" to test

cparrapa, 2021-05-21:
Hi @quangvtvp 

Have you tried all other examples?
Did you follow all the setup steps for the online version here? https://ottoschool.com/en/config-blockly/

Thanks for checking

quangvtvp, 2021-06-05:
Hi @cparrapa 
I still get the same error event I did follow this guide  https://ottoschool.com/en/config-blockly/
I using chrome on macbook pro

cparrapa, 2021-06-06:
Hi @quangvtvp 
Can you try a different example please like the dance?
thanks

cparrapa, 2021-12-09:
Hi @quangvtvp 
This has been fixed please try again and let us know

closed 2021-12-09 by cparrapa