OttoDIY/blockly #107: Erorr when build desktop app for window

quangvtvp, 2021-07-22:
Hi @cparrapa  pls help me check this error

This error occurs when i run command: npm install 
I have aready installed window-build-tools and npm gyp

------------------------------------------

PS D:\Projects\Otto\blockly> npm install

> OttoBlockly@1.0.0 postinstall D:\Projects\Otto\blockly
> electron-builder install-app-deps

Configuring yargs through package.json is deprecated and will be removed in the next major release, please use the JS API instead.
Configuring yargs through package.json is deprecated and will be removed in the next major release, please use the JS API instead.
  • electron-builder version=20.39.0
  • loaded configuration file=package.json ("build" field)
  • rebuilding native production dependencies platform=win32 arch=x64
Error: C:\Program Files\nodejs\node.exe exited with code 1
Output:

> @serialport/bindings@2.0.8 install D:\Projects\Otto\blockly\node_modules\@serialport\bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

D:\Projects\Otto\blockly\node_modules\@serialport\bindings>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" rebuild )
Building the projects in this solution one at a time. To enable parallel build, please add the "-m" switch.
  serialport.cpp
  serialport_win.cpp
  win_delay_load_hook.cc
C:\Users\quanglm2\.electron-gyp\10.1.3\include\node\v8.h(9113,39): warning C4996: 'v8::MicrotasksCompletedCallback': Use *WithData version. (compiling source file ..\src\serialport.cpp) [D:\Projects\Otto\blockly\node_modules\@serialport\bindings\build\bindings.vcxproj]
C:\Users\quanglm2\.electron-gyp\10.1.3\include\node\v8.h(9113,39): warning C4996: 'v8::MicrotasksCompletedCallback': Use *WithData version. (compiling source file ..\src\serialport_win.cpp) [D:\Projects\Otto\blockly\node_modules\@serialport\bindings\build\bindings.vcxproj]
C:\Users\quanglm2\.electron-gyp\10.1.3\include\node\v8.h(9121,42): warning C4996: 'v8::MicrotasksCompletedCallback': Use *WithData version. (compiling source file ..\src\serialport.cpp) [D:\Projects\Otto\blockly\node_modules\@serialport\bindings\build\bindings.vcxproj]
C:\Users\quanglm2\.electron-gyp\10.1.3\include\node\v8.h(9121,42): warning C4996: 'v8::MicrotasksCompletedCallback': Use *WithData version. (compiling source file ..\src\serialport_win.cpp) [D:\Projects\Otto\blockly\node_modules\@serialport\bindings\build\bindings.vcxproj]
D:\Projects\Otto\blockly\node_modules\@serialport\bindings\src\serialport.cpp(329,96): error C2661: 'v8::Object::Set': no overloaded function takes 2 arguments [D:\Projects\Otto\blockly\node_modules\@serialport\bindings\build\bindings.vcxproj]
D:\Projects\Otto\blockly\node_modules\@serialport\bindings\src\serialport.cpp(330,96): error C2661: 'v8::Object::Set': no overloaded function takes 2 arguments [D:\Projects\Otto\blockly\node_modules\@serialport\bindings\build\bindings.vcxproj]
D:\Projects\Otto\blockly\node_modules\@serialport\bindings\src\serialport.cpp(331,96): error C2661: 'v8::Object::Set': no overloaded function takes 2 arguments [D:\Projects\Otto\blockly\node_modules\@serialport\bindings\build\bindings.vcxproj]
D:\Projects\Otto\blockly\node_modules\@serialport\bindings\src\serialport.cpp(378,106): error C2661: 'v8::Object::Set': no overloaded function takes 2 arguments [D:\Projects\Otto\blockly\node_modules\@serialport\bindings\build\bindings.vcxproj]

Error output:
prebuild-install WARN install No prebuilt binaries found (target=10.1.3 runtime=electron arch=x64 libc= platform=win32)
gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Current\Bin\MSBuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:375:28)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:277:12)
gyp ERR! System Windows_NT 10.0.19041
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd D:\Projects\Otto\blockly\node_modules\@serialport\bindings
gyp ERR! node -v v14.17.3
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @serialport/bindings@2.0.8 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the @serialport/bindings@2.0.8 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\quanglm2\AppData\Roaming\npm-cache\_logs\2021-07-22T14_29_38_378Z-debug.log

    at ChildProcess.<anonymous> (D:\Projects\Otto\blockly\node_modules\builder-util\src\util.ts:244:14)
    at Object.onceWrapper (events.js:482:26)
    at ChildProcess.emit (events.js:375:28)
    at maybeClose (internal/child_process.js:1055:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
From previous event:
    at _rebuild (D:\Projects\Otto\blockly\node_modules\app-builder-lib\src\util\yarn.ts:179:5)
    at rebuild (D:\Projects\Otto\blockly\node_modules\app-builder-lib\out\util\yarn.js:163:19)
    at D:\Projects\Otto\blockly\node_modules\app-builder-lib\src\util\yarn.ts:20:11
From previous event:
    at _installOrRebuild (D:\Projects\Otto\blockly\node_modules\app-builder-lib\out\util\yarn.js:71:28)
    at installOrRebuild (D:\Projects\Otto\blockly\node_modules\app-builder-lib\out\util\yarn.js:55:28)
    at D:\Projects\Otto\blockly\node_modules\electron-builder\src\cli\install-app-deps.ts:56:9
    at Generator.next (<anonymous>)
    at processImmediate (internal/timers.js:464:21)
From previous event:
    at _installAppDeps (D:\Projects\Otto\blockly\node_modules\electron-builder\out\cli\install-app-deps.js:176:26)
    at installAppDeps (D:\Projects\Otto\blockly\node_modules\electron-builder\out\cli\install-app-deps.js:144:26)
    at D:\Projects\Otto\blockly\node_modules\electron-builder\src\cli\cli.ts:46:19
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! OttoBlockly@1.0.0 postinstall: `electron-builder install-app-deps`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the OttoBlockly@1.0.0 postinstall script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

quangvtvp, 2021-07-27:
I have avoided the above issue by changing the branch from dependabot/npm_and_yarn/electron-10.1.3
 to master. So now I also have 2 issues when I try to build desktop app for window.

1. First issue: when run : npm start ->. otto blockly screen is shown but usb port disappear
      Step to reproduce these issues:
      1. cd to ../resource/app folder
      2. run : npm install
      3. run: npm run postinstall
      4. run: npm run start. this step shows otto blockly screen
2. the second issue: when run: npm run compiler -> build successful but after install exe file the white screen is shown only
@cparrapa  if you have faced these issues pls help me

cparrapa, 2021-07-28:
@quangvtvp what version are you building?

We just released the version 1.4.3 and it was fine did the same steps as you. I have no idea what those error mean. Post your files so that we can give a try

wtfiAndrewNelson, 2021-10-09:
Please help, when I run the command: npm run compiler It does not compile and I get the error: 

> OttoBlockly@1.4.2 compiler
> build --win --ia32

'build' is not recognized as an internal or external command,
operable program or batch file.

It appears that build is no longer part of npm? Do I need to be using an older version, and if so, which one?