OttoDIY/blockly #111: Touch and Button block bugs

brendonhatcher, 2021-09-08:
OttoBlockly 1.4.3

In the Sensing group is a Touch block.
The block defaults to PIN 2 even though the sensor is attached to A0 (according to the "Getting started with Otto DIY robots" presentation).
It would be simpler for users if it defaulted to A0.

The block contains a checkbox.
I think an empty checkbox means "not pressed" and checked means "pressed".
However, the code is inverted:

Empty checkbox: digitalRead(A0);
Ticked checkbox: (!digitalRead(A0));

The same is true of the Button block.

Finally, the tick in the checkbox is inconsistent.
Sometimes it shows a tick, and sometimes an offset black square.

[image]

cparrapa, 2021-09-08:
Good points @brendonhatcher 
We will update the defaults the got lost in the dropdown conversion of pins.

Thanks for your feedback we will be fixing these stuff.

cparrapa, 2021-09-15:
[image]
Hi @brendonhatcher 
What do you think of something like this?

brendonhatcher, 2021-09-16:
A0 for touch is good.  I don't see a button in the standard build or any of the modular add-ons, so I can't comment on that.
I'm neutral on the finger pressing icon vs the word "pressed".
The checkbox should still be a tick or a blank, because that is more in keeping with standard web form UI.

cparrapa, 2021-09-16:
Hi @brendonhatcher 

Thanks for the feedback, we actually are testing buttons as an alternative to the touch that sometimes is not so reliable. What we want is to have A0 as the pin to swap multiple sensors in Otto so that you can test all kinds of interactions by changing components what do you think?
At least now with the finger press the meaning is now in line with the code since means if the button is on hold or not.
We will updated the check box to tick makes sense

brendonhatcher, 2021-09-16:
That approach sounds good!

cparrapa, 2021-09-22:
fixed here https://github.com/OttoDIY/blockly/commit/e68532c32e563b06385c25dc819042a98d450e55

closed 2021-09-22 by cparrapa