OttoDIY/blockly #113: Build server for Otto Blockly web down?

takujikawata-pr, 2021-09-08:
Hi,

I couldn't upload sketch through Otto Blockly Web.
I had little investigation and it looks to me the build server https://otto-arduino-compiler.mybluemix.net? is now failing.

Here is the screenshot for reference.

[image]

(The body of the server response was
"502 Bad Gateway: Registered endpoint failed to handle the request.")

I'm interested in if there are any way to have my own build server for Otto Blockly Web... If I can, I'll have my own for the backup.

takujikawata-pr, 2021-09-08:
I (reporter of this issue) am closing this issue now because I'm not here reporting code bug but potential server error.
I reported the symptom to the community forum. I think it's more appropriate.

cparrapa, 2021-09-08:
Hi @takujikawata-pr 
It is ok you can still leave the issue here for developers.

Now is possible to upload code to your robot via USB directly from the Chrome browser, but you have to do some extra setup, that we would love to simplify in the future but this is what we got.

1. Download and install Arduino Agent https://create.arduino.cc/getting-started/plugin/welcome
2. Find the config.ing file in the folder where Arduino Agent was installed
On macOS ~/Applications/ArduinoCreateAgent/ArduinoCreateAgent.app/Contents/MacOS/config.ini
On Linux ~/ArduinoCreateAgent/config.ini
On Windows C:\Users\[your user]\AppData\Roaming\ArduinoCreateAgentS
3. Open and modify the origins to "origins = https://ottoschool.com";
4. Save and make sure the name and extension of the file has not been changed "config.ini" config.ini.txt wont work for example
5. Go to [](https://ottoschool.com/blockly)
6. Open the walk Example.
7. Select Arduino Nano old bootloader
8. Connect your Otto and Click in the USB icon
9. Check that Agent Installed: Yes and Agent Connected: Yes
10. Click the upload icon

Wait a bit and your Otto will be walking!.

The **first 5 steps only need to be done one time (setup),** this Beta we will continue working so that the config.ini wont need to be modified.

cparrapa, 2021-12-09:
Hi Takuji 
Thanks we solved now give a try and lets us know

closed 2021-12-09 by cparrapa