OttoDIY/blockly #115: ¿Sería posible añadir esta placa?

Spirit981, 2021-09-18:
Estoy empezando con esto un poco por mi hijo y resulta que tenemos una placa que se compró como Arduino Nano Every, pero en lugar de traer el chip ATMega4809 (que sería el del Every original), trae el AtMega4808. 
He visto mucha gente con este problema y al final en el IDE de Arduino sí que he conseguido añadirla como Thinary Nano Every siguiendo estas instrucciones https://github.com/Thinary/ThinaryArduino

¿Sería muy complicado añadir esta placa al Otto Blocky?

Muchas gracias

cparrapa, 2021-09-19:
Hola @Spirit981 

Y has probado subir código con el Arduino Every seleccionado?
Es complicado porque no tengo una para testear,  donde la compraste?
o lo que puedes hacer es ensayar cambiando la configuración de la placa en tu instalación en este archivo
https://github.com/OttoDIY/blockly/blob/master/www/js/boards.js

Spirit981, 2021-09-19:
Gracias por la rápida respuesta @cparrapa .

Ya intenté seleccionando la Every, tanto en Otto Blocky como en IDE arduino, pero da error.
También había intentando modificar ese archivo, pero por alguna razón cuando abro Otto Blocky, no me muestra la mayoría de elementos (algo muy raro).

La placa la compré en aliexpress, pero después de investigar un poco he visto que este problema le está pasando a más gente y la solución para trabajar con ella en el IDE es la que menciono en el primer post, pero aún así hay librerías que no las reconoce por ser avr.

No te preocupes, esta creo que la dejaré por un cajón guardada y a ver si puedo esta semana comprar un nano “normal” y se acaban los problemas.

Lo dicho, muchas gracias por la respuesta.

closed 2021-09-19 by cparrapa