OttoDIY/blockly #117: Starting the serial monitor resets the value of variables

brendonhatcher, 2021-09-21:
I created the attached code.
I downloaded it to Otto.
Tapping on the sensor shows the number "1" and repeat tapping increments the number by 1 each time.
I then open the Serial Monitor and press "Start".
When I tap again, the number displayed is "1".
I continue to tap, and the number increments.
I then close the Serial Monitor and tap again.
The number has reset to 1 again

[image]
[Otto_block_tap to count.zip](https://github.com/OttoDIY/blockly/files/7202728/Otto_block_tap.to.count.zip)

cparrapa, 2021-09-21:
Hi @brendonhatcher 

I think is because the serial monitor resets the board therefore start form the beginning again every time you close, I do not reacll if in arduino IDE it is same situation actually

brendonhatcher, 2021-09-22:
I found this post that confirms that the issue is the same for the Arduino IDE.  It appears that the board resets as a result of the configuration of the Windows (in my case) software providing the serial communication between the Nano and OttoBlockly.  This is either a simple code change in OttoBlockly, or something too complex but insignificant to try to fix.

https://forum.arduino.cc/t/serial-connect-resets-board/25409

cparrapa, 2021-10-05:
Make sense is far beyond the actual Arduino IDE, we can add a warning when explaining the concept of serial monitor. A way to store the variable will be to use the EEPROM memory

closed 2021-10-30 by cparrapa