OttoDIY/blockly #118: Otto movements without decalring pins

brendonhatcher, 2021-09-21:
I understand logically that you cannot use Home or Walk blocks without declaring the pins in the setup.
However, my users won't understand this intuitively.
The arduino error that comes up if you try won't help them either.

Is there any way that block dependencies could be flagged?
So, if you drag in "Home" without "Pins" you get a warning?

[image]

cparrapa, 2021-09-21:
Hi @brendonhatcher 

This was a main dilemma since many  users requested to have control of the pins, then it came with a trade off which was having to always declared the pins in the setup. I think there is a way to add a warning but i tried last time and it did not work, we need some help here from blockly experts.

brendonhatcher, 2021-09-22:
Would it be possible to have default pin settings that don't require the pin block.
Then, if someone drags in a pin block, the settings that they add there overwrite the defaults.
I think that the vast majority of users would have set their pins according to the build manual, and therefore don't require control of the pins.

cparrapa, 2021-09-22:
Right @brendonhatcher but also at the same time it is a more smooth transition to real coding in Arduino were you have to make sure to define your pins.

I like your approach i just don't see an easy way to have that behavioural conditions in the blocks they are very straight with the code generation.

brendonhatcher, 2021-09-23:
I agree with your comment that defining your pins is part of the transition to "real" coding.  I'm going to check that there are blocks for all the common sensors and actuators in Otto, Otto plus and the modular add-ons.

cparrapa, 2021-10-05:
Cool @brendonhatcher we will just make sure to explain this in the guide more in detail for beginners, in this way they also learn and advance users can modify default pins.
Any other block missing with this setups that you considered important?

closed 2021-10-07 by cparrapa