OttoDIY/blockly #119: Unattached blocks still function

brendonhatcher, 2021-09-21:
I think I have raised this before, but the current approach is deeply disturbing in terms of teaching computational thinking.

Unattached blocs function as if they were attached.
This is not only poor logic, but is also contrary to apps such as Scratch.

It also prevents a very important function...
I have a stack of blocks which are not functioning correctly.
I disconnect parts of the stack to identify where the problem is.
In scratch that works.  In Ottoblockly, the disconnected blocks still function, but not in the order that I would expect.
The only way to prevent them from functioning is to delete them entirely, but I don't want to have to rebuild them later.

[image]

brendonhatcher, 2021-09-22:
Possible solution? https://groups.google.com/g/blockly/c/0LEZTdD4Bq8/m/h9-QCl2UAAAJ

cparrapa, 2021-09-22:
thanks for helping us find the solution it seems is there so time to learn how to implement. Wish me luck