OttoDIY/blockly #120: Variable block language issue in English

brendonhatcher, 2021-09-21:
I think that there is a language translation issue in variables.
When I create a variable, the blocks don't make perfect sense in English, and I wonder if they were translated literally from another language?

e.g. "put the variable xxxx at" should probably be "set the variable xxx to"

[image]

cparrapa, 2021-09-22:
Hi @brendonhatcher 
Indeed these blocks have been a nightmare also for us.
I am thinking then to use more simple English How about like this?
[image]

brendonhatcher, 2021-09-23:
I'll have a look at what each does and then suggest some text changes.

cparrapa, 2021-10-05:
Thanks let me know and we add as you request @brendonhatcher

cparrapa, 2021-10-07:
how about  like this ? @brendonhatcher 
[image]

closed 2021-10-30 by cparrapa