OttoDIY/blockly #122: Otto Expansion Humanoid setup and blocks

brendonhatcher, 2021-09-29:
There are a few gaps that make the humanoid expansion hard to use:

1. No installation manual - I had to find the pins by adding the block and then previewing the arduino code.
2. Not included in the calibration code or app
3. Inconsistent blocks - the legs require the pins to be defined using a block in setup, but the arm block has the pins hardcoded
4. There is no "Home" block, or option for "home" for the arms

cparrapa, 2021-10-05:
Hi @Pawka 

1. Manual in progress check the guide here and let me know what you think https://docs.google.com/presentation/d/1VRzc4KMxjN2FYakP-p0wuHSYh6-RGmXOqKiz24fdLck/edit#slide=id.gf5d44e4b6b_0_62
2. The arms are not so sensitive to the calibration if you assemble in the center. Do you really need to calibrate them?
3. We will add block for the arms pins setup
4. Good idea we can add that option 

Thanks

cparrapa, 2021-10-30:
[image]

@brendonhatcher like this?

cparrapa, 2021-11-09:
solved for https://github.com/OttoDIY/blockly/releases/tag/v1.4.4

closed 2021-11-09 by cparrapa