OttoDIY/blockly #134: error on compiled app for linux - no usb option and app freeze

elton-bt, 2021-11-05:
Hi, when testing the app with npm start, everything works. I can [test](https://pasteboard.co/dF4qh4V2qkzD.png) the code and [upload](https://pasteboard.co/tXW7lpSehrOI.png) it to otto.

However, when I run npm run compiler, in the generated .appImage, when executed, doesn't appear the [option](https://pasteboard.co/azytUKecg2Qa.png) to choose the usb port. Also, I can't [check](https://pasteboard.co/kLYiJT8DaVlj.png) the code and the app freeze when i try.

cparrapa, 2021-11-17:
Hi @elton-bt 
Linux version is in this other repo https://github.com/OttoDIY/blocklyLinux

We are planning to merge onto this but we need help on that

elton-bt, 2021-11-18:
Hi @cparrapa . When I opened the issue I was testing the linux version present here: https://github.com/OttoDIY/blockly/tree/versionlinux. Here, npm start opens the app that works. But the .appimage generated with the npm run compiler is having problems as reported by me earlier.

As you sent me the new link (https://github.com/OttoDIY/blocklyLinux) I did all the procedure to build the project again. Now the app opened by npm start [doesn't work](https://pasteboard.co/eTFSxzAFIVDy.png) and the .appimage generated with the npm run compiler [doesn't work](https://pasteboard.co/ulOzJmIVcLUY.png) either, just like before. I installed the .deb and the same error remains: It doesn't allow the choice of usb ports, and when verifying the code the application crashes.

zelogik, 2021-12-27:
Same here, the 1.4.2 AppImage on the release page don't allow to change USB and freeze completely when try to compile/upload.

And the latest version from github work with npm start (when you change fs/promises bug) but have the same behavior as the compiled 1.4.2 AppImage.

elton-bt, 2021-12-27:
@zelogik , the same problem that I mentioned. But, unfortunately, the issue was closed. @cparrapa can you reopen the issue?

cparrapa, 2022-01-11:
Now is reopen @elton-bt

closed 2024-06-30 by cparrapa