OttoDIY/blockly #138: apds9960 gesture sensor block

linsThankachan, 2021-12-02:
the block for apds9960 gesture sensor is not working. there is "gesture enable" block which is not enabling the gesture but disabling it 
and also there is problem for color detection also same problem

[image]

for gesture to work there should be a proximity enable also
[image]
.

cparrapa, 2021-12-05:
Great Find @linsThankachan what sensor do you use?
I constantly find issues with this sensors even on the hardware part

linsThankachan, 2021-12-05:
it is a generic apds9960 sensor. you can see that whenever we do enable. the code side will be disabled.

linsThankachan, 2021-12-05:
hello, i have checked with sparkfun apds library it is not working but adafruit apds9960 library it is working. ottoblockly uses sparkfun's library

cparrapa, 2021-12-06:
Great Find thanks!  @linsThankachan  we will make sure to do this update for next release. 
Any other issue you have seen?

linsThankachan, 2021-12-07:
the ir reciver code for remote also not working.

linsThankachan, 2021-12-07:
when will be the next release because we are using ottoblockly for our products.

logix5, 2021-12-08:
Hi lins

Can you show me your code using this sensor? I have tested previously in my
blockly and the blocks worked. Maybe the library that i used was updated

thanks in advance

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Libre
de virus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

-- 
Óscar Ferruz Romero
*Socio fundador.*
*Tel. 620.402.168*

 C/ Artajona, 17.  28039 - Madrid.
www.logix5.com

www.twitter.com/Logix5_SS <http://www.twitter.com/logix5_SS>

www.facebook.com/logix5

www.instagram.com/logix5_smart_solutions

Conforme dispone el Reglamento (UE) 2016/679 del Parlamento Europeo y del
Consejo, de 27 de abril de 2016, relativo a la protección de las personas
físicas en lo referente al tratamiento de datos personales y a la libre
circulación de estos, le informamos que los datos personales facilitados y
recogidos en este e-mail y sus documentos adjuntos serán tratados e
incluidos automatizadamente en los ficheros de datos personales bajo
responsabilidad de LOGIX5 SMART SOLUTIONS, S.L. B-86129699, donde se
recogen y almacenan los datos personales con la finalidad exclusiva de
formar parte en los procesos de selección de personal, bolsa de trabajo,
contratación o gestiones relacionadas con temas laborales que se lleven a
cabo. El interesado otorga su consentimiento para el tratamiento de sus
datos personales con la finalidad anteriormente mencionada. En el supuesto
de producirse alguna modificación de sus datos personales, le solicitamos,
nos lo comunique por escrito con la única finalidad de mantenerlos
actualizados.   LOGIX5 SMART SOLUTIONS, S.L. B-86129699 se compromete a
respetar su confidencialidad en el tratamiento de sus datos personales, y
le informa que tiene derecho a ejercitar los derechos ARCO (derecho de
acceso, rectificación, cancelación y oposición) de sus datos personales
mediante solicitud escrita a ***@***.***, adjuntando fotocopia del
D.N.I.

"En la vida hay tres cosas que no dan marcha atrás: la palabra pronunciada,
la flecha lanzada y la oportunidad perdida, ¿vamos a dejarla escapar?"

linsThankachan, 2021-12-08:
[image]
the above is  code in ottoblockly . it is not working.

i have given below the working code using arduino ide and using adafruit apds library. i have checked and sparkfun's apds library is not working. given below is the example code for gesture sensing.
`/***************************************************************************
  This is a library for the APDS9960 digital proximity, ambient light, RGB, and gesture sensor

  This sketch puts the sensor in gesture mode and decodes gestures.
  To use this, first put your hand close to the sensor to enable gesture mode.
  Then move your hand about 6" from the sensor in the up -> down, down -> up, 
  left -> right, or right -> left direction.

  Designed specifically to work with the Adafruit APDS9960 breakout
  ----> http://www.adafruit.com/products/3595

  These sensors use I2C to communicate. The device's I2C address is 0x39

  Adafruit invests time and resources providing this open source code,
  please support Adafruit andopen-source hardware by purchasing products
  from Adafruit!

  Written by Dean Miller for Adafruit Industries.
  BSD license, all text above must be included in any redistribution
 ***************************************************************************/

#include "Adafruit_APDS9960.h"
Adafruit_APDS9960 apds;

// the setup function runs once when you press reset or power the board
void setup() {
  Serial.begin(115200);
  
  if(!apds.begin()){
    Serial.println("failed to initialize device! Please check your wiring.");
  }
  else Serial.println("Device initialized!");

  //gesture mode will be entered once proximity mode senses something close
  apds.enableProximity(true);
  apds.enableGesture(true);
}

// the loop function runs over and over again forever
void loop() {
  //read a gesture from the device
    uint8_t gesture = apds.readGesture();
    if(gesture == APDS9960_DOWN) Serial.println("v");
    if(gesture == APDS9960_UP) Serial.println("^");
    if(gesture == APDS9960_LEFT) Serial.println("<");
    if(gesture == APDS9960_RIGHT) Serial.println(">");
}`

logix5, 2021-12-08:
Hi ,

the program to test is similar than this:
[image: image.png]

This code don´t use the interrupts, but in arduino leonardo i tested years
ago and works perfectly

Let us know

-- 
Óscar Ferruz Romero
*Socio fundador.*
*Tel. 620.402.168*

 C/ Artajona, 17.  28039 - Madrid.
www.logix5.com

www.twitter.com/Logix5_SS <http://www.twitter.com/logix5_SS>

www.facebook.com/logix5

www.instagram.com/logix5_smart_solutions

Conforme dispone el Reglamento (UE) 2016/679 del Parlamento Europeo y del
Consejo, de 27 de abril de 2016, relativo a la protección de las personas
físicas en lo referente al tratamiento de datos personales y a la libre
circulación de estos, le informamos que los datos personales facilitados y
recogidos en este e-mail y sus documentos adjuntos serán tratados e
incluidos automatizadamente en los ficheros de datos personales bajo
responsabilidad de LOGIX5 SMART SOLUTIONS, S.L. B-86129699, donde se
recogen y almacenan los datos personales con la finalidad exclusiva de
formar parte en los procesos de selección de personal, bolsa de trabajo,
contratación o gestiones relacionadas con temas laborales que se lleven a
cabo. El interesado otorga su consentimiento para el tratamiento de sus
datos personales con la finalidad anteriormente mencionada. En el supuesto
de producirse alguna modificación de sus datos personales, le solicitamos,
nos lo comunique por escrito con la única finalidad de mantenerlos
actualizados.   LOGIX5 SMART SOLUTIONS, S.L. B-86129699 se compromete a
respetar su confidencialidad en el tratamiento de sus datos personales, y
le informa que tiene derecho a ejercitar los derechos ARCO (derecho de
acceso, rectificación, cancelación y oposición) de sus datos personales
mediante solicitud escrita a ***@***.***, adjuntando fotocopia del
D.N.I.

"En la vida hay tres cosas que no dan marcha atrás: la palabra pronunciada,
la flecha lanzada y la oportunidad perdida, ¿vamos a dejarla escapar?"

linsThankachan, 2021-12-09:
can you give that code so that i can test it.
In ottoblockly also there is no interrupt correct.

cparrapa, 2021-12-09:
> the ir receiver code for remote also not working.

Hi @linsThankachan 
Did you use this example https://github.com/OttoDIY/blockly/issues/128 and what IR receiver are you using?
What remote control and microcontroller board are you using?

logix5, 2021-12-09:
Hi , the example code of these blocks is:

[image: image.png]

Check and let us know

BR

-- 
Óscar Ferruz Romero
*Socio fundador.*
*Tel. 620.402.168*

 C/ Artajona, 17.  28039 - Madrid.
www.logix5.com

www.twitter.com/Logix5_SS <http://www.twitter.com/logix5_SS>

www.facebook.com/logix5

www.instagram.com/logix5_smart_solutions

Conforme dispone el Reglamento (UE) 2016/679 del Parlamento Europeo y del
Consejo, de 27 de abril de 2016, relativo a la protección de las personas
físicas en lo referente al tratamiento de datos personales y a la libre
circulación de estos, le informamos que los datos personales facilitados y
recogidos en este e-mail y sus documentos adjuntos serán tratados e
incluidos automatizadamente en los ficheros de datos personales bajo
responsabilidad de LOGIX5 SMART SOLUTIONS, S.L. B-86129699, donde se
recogen y almacenan los datos personales con la finalidad exclusiva de
formar parte en los procesos de selección de personal, bolsa de trabajo,
contratación o gestiones relacionadas con temas laborales que se lleven a
cabo. El interesado otorga su consentimiento para el tratamiento de sus
datos personales con la finalidad anteriormente mencionada. En el supuesto
de producirse alguna modificación de sus datos personales, le solicitamos,
nos lo comunique por escrito con la única finalidad de mantenerlos
actualizados.   LOGIX5 SMART SOLUTIONS, S.L. B-86129699 se compromete a
respetar su confidencialidad en el tratamiento de sus datos personales, y
le informa que tiene derecho a ejercitar los derechos ARCO (derecho de
acceso, rectificación, cancelación y oposición) de sus datos personales
mediante solicitud escrita a ***@***.***, adjuntando fotocopia del
D.N.I.

"En la vida hay tres cosas que no dan marcha atrás: la palabra pronunciada,
la flecha lanzada y la oportunidad perdida, ¿vamos a dejarla escapar?"

linsThankachan, 2021-12-10:
@logix5  cannot see your image

logix5, 2021-12-10:
[image: image.png]

Try now

BR

-- 
Óscar Ferruz Romero
*Socio fundador.*
*Tel. 620.402.168*

 C/ Artajona, 17.  28039 - Madrid.
www.logix5.com

www.twitter.com/Logix5_SS <http://www.twitter.com/logix5_SS>

www.facebook.com/logix5

www.instagram.com/logix5_smart_solutions

Conforme dispone el Reglamento (UE) 2016/679 del Parlamento Europeo y del
Consejo, de 27 de abril de 2016, relativo a la protección de las personas
físicas en lo referente al tratamiento de datos personales y a la libre
circulación de estos, le informamos que los datos personales facilitados y
recogidos en este e-mail y sus documentos adjuntos serán tratados e
incluidos automatizadamente en los ficheros de datos personales bajo
responsabilidad de LOGIX5 SMART SOLUTIONS, S.L. B-86129699, donde se
recogen y almacenan los datos personales con la finalidad exclusiva de
formar parte en los procesos de selección de personal, bolsa de trabajo,
contratación o gestiones relacionadas con temas laborales que se lleven a
cabo. El interesado otorga su consentimiento para el tratamiento de sus
datos personales con la finalidad anteriormente mencionada. En el supuesto
de producirse alguna modificación de sus datos personales, le solicitamos,
nos lo comunique por escrito con la única finalidad de mantenerlos
actualizados.   LOGIX5 SMART SOLUTIONS, S.L. B-86129699 se compromete a
respetar su confidencialidad en el tratamiento de sus datos personales, y
le informa que tiene derecho a ejercitar los derechos ARCO (derecho de
acceso, rectificación, cancelación y oposición) de sus datos personales
mediante solicitud escrita a ***@***.***, adjuntando fotocopia del
D.N.I.

"En la vida hay tres cosas que no dan marcha atrás: la palabra pronunciada,
la flecha lanzada y la oportunidad perdida, ¿vamos a dejarla escapar?"

linsThankachan, 2021-12-10:
cannot see it. it is showing only text

linsThankachan, 2021-12-10:
[image]

logix5, 2021-12-10:
Hi

This is the code that these block are generating:

#include <SparkFun_APDS9960.h>

SparkFun_APDS9960 apds = SparkFun_APDS9960();

void setup() {
  Serial.begin(115200);

  apds.init();

    apds.enableGestureSensor(false);

}

void loop() {
    if (apds.isGestureAvailable()) {
      Serial.println(apds.readGesture());
    }

}

-- 
Óscar Ferruz Romero
*Socio fundador.*
*Tel. 620.402.168*

 C/ Artajona, 17.  28039 - Madrid.
www.logix5.com

www.twitter.com/Logix5_SS <http://www.twitter.com/logix5_SS>

www.facebook.com/logix5

www.instagram.com/logix5_smart_solutions

Conforme dispone el Reglamento (UE) 2016/679 del Parlamento Europeo y del
Consejo, de 27 de abril de 2016, relativo a la protección de las personas
físicas en lo referente al tratamiento de datos personales y a la libre
circulación de estos, le informamos que los datos personales facilitados y
recogidos en este e-mail y sus documentos adjuntos serán tratados e
incluidos automatizadamente en los ficheros de datos personales bajo
responsabilidad de LOGIX5 SMART SOLUTIONS, S.L. B-86129699, donde se
recogen y almacenan los datos personales con la finalidad exclusiva de
formar parte en los procesos de selección de personal, bolsa de trabajo,
contratación o gestiones relacionadas con temas laborales que se lleven a
cabo. El interesado otorga su consentimiento para el tratamiento de sus
datos personales con la finalidad anteriormente mencionada. En el supuesto
de producirse alguna modificación de sus datos personales, le solicitamos,
nos lo comunique por escrito con la única finalidad de mantenerlos
actualizados.   LOGIX5 SMART SOLUTIONS, S.L. B-86129699 se compromete a
respetar su confidencialidad en el tratamiento de sus datos personales, y
le informa que tiene derecho a ejercitar los derechos ARCO (derecho de
acceso, rectificación, cancelación y oposición) de sus datos personales
mediante solicitud escrita a ***@***.***, adjuntando fotocopia del
D.N.I.

"En la vida hay tres cosas que no dan marcha atrás: la palabra pronunciada,
la flecha lanzada y la oportunidad perdida, ¿vamos a dejarla escapar?"

linsThankachan, 2021-12-10:
[image]
the above is the block code im uploaded. it is the same as you have given. but it is not working. I have checked in arduino ide also with same sparfun library but it is not working. but when i use adafruit apds9960 library it is working.

logix5, 2021-12-10:
Ok i have tested this module in leonardo board and worked, but maybe any
update of the library is necessary. Please i have this advice to you:

Using the Arduino IDE, install the library, and execute an example using
this library. If this example works then the problem are our blocks else
maybe the problem is the sensor

Let us know

-- 
Óscar Ferruz Romero
*Socio fundador.*
*Tel. 620.402.168*

 C/ Artajona, 17.  28039 - Madrid.
www.logix5.com

www.twitter.com/Logix5_SS <http://www.twitter.com/logix5_SS>

www.facebook.com/logix5

www.instagram.com/logix5_smart_solutions

Conforme dispone el Reglamento (UE) 2016/679 del Parlamento Europeo y del
Consejo, de 27 de abril de 2016, relativo a la protección de las personas
físicas en lo referente al tratamiento de datos personales y a la libre
circulación de estos, le informamos que los datos personales facilitados y
recogidos en este e-mail y sus documentos adjuntos serán tratados e
incluidos automatizadamente en los ficheros de datos personales bajo
responsabilidad de LOGIX5 SMART SOLUTIONS, S.L. B-86129699, donde se
recogen y almacenan los datos personales con la finalidad exclusiva de
formar parte en los procesos de selección de personal, bolsa de trabajo,
contratación o gestiones relacionadas con temas laborales que se lleven a
cabo. El interesado otorga su consentimiento para el tratamiento de sus
datos personales con la finalidad anteriormente mencionada. En el supuesto
de producirse alguna modificación de sus datos personales, le solicitamos,
nos lo comunique por escrito con la única finalidad de mantenerlos
actualizados.   LOGIX5 SMART SOLUTIONS, S.L. B-86129699 se compromete a
respetar su confidencialidad en el tratamiento de sus datos personales, y
le informa que tiene derecho a ejercitar los derechos ARCO (derecho de
acceso, rectificación, cancelación y oposición) de sus datos personales
mediante solicitud escrita a ***@***.***, adjuntando fotocopia del
D.N.I.

"En la vida hay tres cosas que no dan marcha atrás: la palabra pronunciada,
la flecha lanzada y la oportunidad perdida, ¿vamos a dejarla escapar?"

linsThankachan, 2021-12-10:
I have already tested the sensor with sparkfun library which i have installed 5 or 6 days before which means it is updated, for your consideration i have checked one more time also there is was no updates.

ottoblockly doesn't have an interrupt in the code side but sparkfun's example has it . so first i tried with no interrupt there was no result. then i have connected with an interrupt pin that the example pointed then also no output.

The sensor has no problem because it is working perfectly with adafruit apds9960 library which i used and got the output .

please check if your library is updated.
please download adafruit library and test that also if that is working we can change to adafruit library.

i'am using an arduino uno

i don't know how you are getting the output from blockly because the boolean that enables the gesture sensor is false when we give true in blocks.

logix5, 2021-12-10:
Thanks, I have been reading examples but all are using interrupts.

Please can you test these options:

a) the same program but you don't connect the IR pin

b) Use the same program but using apds.enableGestureSensor(true); in the
Arduino IDE. Test connecting and disconnecting the interrupt pin

Let us know. I don´t know when I could test the sensor again. Few years ago
this code was working but i don't remember my connection.

I am going to surf and try to look for an example using polling mode,
without interrupt

BR

-- 
Óscar Ferruz Romero
*Socio fundador.*
*Tel. 620.402.168*

 C/ Artajona, 17.  28039 - Madrid.
www.logix5.com

www.twitter.com/Logix5_SS <http://www.twitter.com/logix5_SS>

www.facebook.com/logix5

www.instagram.com/logix5_smart_solutions

Conforme dispone el Reglamento (UE) 2016/679 del Parlamento Europeo y del
Consejo, de 27 de abril de 2016, relativo a la protección de las personas
físicas en lo referente al tratamiento de datos personales y a la libre
circulación de estos, le informamos que los datos personales facilitados y
recogidos en este e-mail y sus documentos adjuntos serán tratados e
incluidos automatizadamente en los ficheros de datos personales bajo
responsabilidad de LOGIX5 SMART SOLUTIONS, S.L. B-86129699, donde se
recogen y almacenan los datos personales con la finalidad exclusiva de
formar parte en los procesos de selección de personal, bolsa de trabajo,
contratación o gestiones relacionadas con temas laborales que se lleven a
cabo. El interesado otorga su consentimiento para el tratamiento de sus
datos personales con la finalidad anteriormente mencionada. En el supuesto
de producirse alguna modificación de sus datos personales, le solicitamos,
nos lo comunique por escrito con la única finalidad de mantenerlos
actualizados.   LOGIX5 SMART SOLUTIONS, S.L. B-86129699 se compromete a
respetar su confidencialidad en el tratamiento de sus datos personales, y
le informa que tiene derecho a ejercitar los derechos ARCO (derecho de
acceso, rectificación, cancelación y oposición) de sus datos personales
mediante solicitud escrita a ***@***.***, adjuntando fotocopia del
D.N.I.

"En la vida hay tres cosas que no dan marcha atrás: la palabra pronunciada,
la flecha lanzada y la oportunidad perdida, ¿vamos a dejarla escapar?"

linsThankachan, 2021-12-10:
The adafruit library doesn't need interrupt.
I have tried what you are telling now. But it will not work with the
sparkfun library.

So please use adafruit apds9960 library

logix5, 2021-12-10:
Ok thanks i will review it and i will change the code in this case

BR

-- 
Óscar Ferruz Romero
*Socio fundador.*
*Tel. 620.402.168*

 C/ Artajona, 17.  28039 - Madrid.
www.logix5.com

www.twitter.com/Logix5_SS <http://www.twitter.com/logix5_SS>

www.facebook.com/logix5

www.instagram.com/logix5_smart_solutions

Conforme dispone el Reglamento (UE) 2016/679 del Parlamento Europeo y del
Consejo, de 27 de abril de 2016, relativo a la protección de las personas
físicas en lo referente al tratamiento de datos personales y a la libre
circulación de estos, le informamos que los datos personales facilitados y
recogidos en este e-mail y sus documentos adjuntos serán tratados e
incluidos automatizadamente en los ficheros de datos personales bajo
responsabilidad de LOGIX5 SMART SOLUTIONS, S.L. B-86129699, donde se
recogen y almacenan los datos personales con la finalidad exclusiva de
formar parte en los procesos de selección de personal, bolsa de trabajo,
contratación o gestiones relacionadas con temas laborales que se lleven a
cabo. El interesado otorga su consentimiento para el tratamiento de sus
datos personales con la finalidad anteriormente mencionada. En el supuesto
de producirse alguna modificación de sus datos personales, le solicitamos,
nos lo comunique por escrito con la única finalidad de mantenerlos
actualizados.   LOGIX5 SMART SOLUTIONS, S.L. B-86129699 se compromete a
respetar su confidencialidad en el tratamiento de sus datos personales, y
le informa que tiene derecho a ejercitar los derechos ARCO (derecho de
acceso, rectificación, cancelación y oposición) de sus datos personales
mediante solicitud escrita a ***@***.***, adjuntando fotocopia del
D.N.I.

"En la vida hay tres cosas que no dan marcha atrás: la palabra pronunciada,
la flecha lanzada y la oportunidad perdida, ¿vamos a dejarla escapar?"

linsThankachan, 2021-12-11:
ok thanks

linsThankachan, 2021-12-11:
hello i want to ask one thing which is not related to this one.
is otto scratch ai opensource?

logix5, 2021-12-11:
Hi , I don´t know, I think the answer is no, but Camilo can explain
better.  I would like the option to add more blocks features in this
software because it is very interesting to manage different boards too

BR

-- 
Óscar Ferruz Romero
*Socio fundador.*
*Tel. 620.402.168*

 C/ Artajona, 17.  28039 - Madrid.
www.logix5.com

www.twitter.com/Logix5_SS <http://www.twitter.com/logix5_SS>

www.facebook.com/logix5

www.instagram.com/logix5_smart_solutions

Conforme dispone el Reglamento (UE) 2016/679 del Parlamento Europeo y del
Consejo, de 27 de abril de 2016, relativo a la protección de las personas
físicas en lo referente al tratamiento de datos personales y a la libre
circulación de estos, le informamos que los datos personales facilitados y
recogidos en este e-mail y sus documentos adjuntos serán tratados e
incluidos automatizadamente en los ficheros de datos personales bajo
responsabilidad de LOGIX5 SMART SOLUTIONS, S.L. B-86129699, donde se
recogen y almacenan los datos personales con la finalidad exclusiva de
formar parte en los procesos de selección de personal, bolsa de trabajo,
contratación o gestiones relacionadas con temas laborales que se lleven a
cabo. El interesado otorga su consentimiento para el tratamiento de sus
datos personales con la finalidad anteriormente mencionada. En el supuesto
de producirse alguna modificación de sus datos personales, le solicitamos,
nos lo comunique por escrito con la única finalidad de mantenerlos
actualizados.   LOGIX5 SMART SOLUTIONS, S.L. B-86129699 se compromete a
respetar su confidencialidad en el tratamiento de sus datos personales, y
le informa que tiene derecho a ejercitar los derechos ARCO (derecho de
acceso, rectificación, cancelación y oposición) de sus datos personales
mediante solicitud escrita a ***@***.***, adjuntando fotocopia del
D.N.I.

"En la vida hay tres cosas que no dan marcha atrás: la palabra pronunciada,
la flecha lanzada y la oportunidad perdida, ¿vamos a dejarla escapar?"

logix5, 2021-12-11:
Hi again

I have uploaded the new block and library to the repository. I can´t test
because I don't have the sensor now. Please can you update and test it.

I have changed the simple gesture and color  blocks, but maybe another
blocks can be useful to add if you need. I have updated for arduino and
ESP8266 boards

Let us know
BR

-- 
Óscar Ferruz Romero
*Socio fundador.*
*Tel. 620.402.168*

 C/ Artajona, 17.  28039 - Madrid.
www.logix5.com

www.twitter.com/Logix5_SS <http://www.twitter.com/logix5_SS>

www.facebook.com/logix5

www.instagram.com/logix5_smart_solutions

Conforme dispone el Reglamento (UE) 2016/679 del Parlamento Europeo y del
Consejo, de 27 de abril de 2016, relativo a la protección de las personas
físicas en lo referente al tratamiento de datos personales y a la libre
circulación de estos, le informamos que los datos personales facilitados y
recogidos en este e-mail y sus documentos adjuntos serán tratados e
incluidos automatizadamente en los ficheros de datos personales bajo
responsabilidad de LOGIX5 SMART SOLUTIONS, S.L. B-86129699, donde se
recogen y almacenan los datos personales con la finalidad exclusiva de
formar parte en los procesos de selección de personal, bolsa de trabajo,
contratación o gestiones relacionadas con temas laborales que se lleven a
cabo. El interesado otorga su consentimiento para el tratamiento de sus
datos personales con la finalidad anteriormente mencionada. En el supuesto
de producirse alguna modificación de sus datos personales, le solicitamos,
nos lo comunique por escrito con la única finalidad de mantenerlos
actualizados.   LOGIX5 SMART SOLUTIONS, S.L. B-86129699 se compromete a
respetar su confidencialidad en el tratamiento de sus datos personales, y
le informa que tiene derecho a ejercitar los derechos ARCO (derecho de
acceso, rectificación, cancelación y oposición) de sus datos personales
mediante solicitud escrita a ***@***.***, adjuntando fotocopia del
D.N.I.

"En la vida hay tres cosas que no dan marcha atrás: la palabra pronunciada,
la flecha lanzada y la oportunidad perdida, ¿vamos a dejarla escapar?"

linsThankachan, 2021-12-13:
hi ,
i have tested it . it is working in the online editor.
thanks you very much.

closed 2021-12-19 by BodoMinea