OttoDIY/blockly #139: ¿Escornabot?

NeftaliVenancio, 2021-12-16:
Si no mal recuerdo había una categoría de bloques dedicada al escornabot. ¿Hay planes de integrar bloques relacionados nuevamente?

BodoMinea, 2021-12-19:
It's best to write issues in English.

I see that the code regarding the escornabot is still here - https://github.com/OttoDIY/blockly/blob/master/www/blocs%26generateurs/escornabot.js

BodoMinea, 2021-12-19:
I haven't worked on the project in a while and don't know why the blocks won't show up. Last time I used Blockly there was an option to select block visibility based on a category checklist, I see that's gone and replaced with the three complexity levels in the menu bar. 

Maybe the blocks you are looking for disappeared during this change somehow?

BodoMinea, 2021-12-19:
At the very basic level it seems that the Escornabot blocks are not shown because they are not present in the XML loaded from https://github.com/OttoDIY/blockly/tree/master/www/toolbox

(toolbox_master.xml has them but doesn't have the „level” based separation, toolbox_arduino_all.xml seems to be the default one and is completely missing these blocks)

I don't know if that's because some recently arisen compatibility issue, I'll let someone more familiar with the project's recent development look into it.

cparrapa, 2021-12-20:
Hi @NeftaliVenancio  @BodoMinea 
Yes we forgot to unhide it from the toolbar in the next release we will add again for level 3
Thanks

cparrapa, 2022-01-11:
https://github.com/OttoDIY/blockly/commits/master

closed 2022-01-11 by cparrapa