OttoDIY/blockly #146: Block for multiple I2C

cparrapa, 2022-01-24:
Allow I2c bus for multiple I2c items like OLED and RTC, TOF sensors

I2C expander can I run multiple I2C

logix5, 2022-01-25:
Hi ,  do you refer to a hardware solution? right?

BR

-- 
Óscar Ferruz Romero
*Socio fundador.*
*Tel. 620.402.168*

 C/ Artajona, 17.  28039 - Madrid.
www.logix5.com

www.twitter.com/Logix5_SS <http://www.twitter.com/logix5_SS>

www.facebook.com/logix5

www.instagram.com/logix5_smart_solutions

Conforme dispone el Reglamento (UE) 2016/679 del Parlamento Europeo y del
Consejo, de 27 de abril de 2016, relativo a la protección de las personas
físicas en lo referente al tratamiento de datos personales y a la libre
circulación de estos, le informamos que los datos personales facilitados y
recogidos en este e-mail y sus documentos adjuntos serán tratados e
incluidos automatizadamente en los ficheros de datos personales bajo
responsabilidad de LOGIX5 SMART SOLUTIONS, S.L. B-86129699, donde se
recogen y almacenan los datos personales con la finalidad exclusiva de
formar parte en los procesos de selección de personal, bolsa de trabajo,
contratación o gestiones relacionadas con temas laborales que se lleven a
cabo. El interesado otorga su consentimiento para el tratamiento de sus
datos personales con la finalidad anteriormente mencionada. En el supuesto
de producirse alguna modificación de sus datos personales, le solicitamos,
nos lo comunique por escrito con la única finalidad de mantenerlos
actualizados.   LOGIX5 SMART SOLUTIONS, S.L. B-86129699 se compromete a
respetar su confidencialidad en el tratamiento de sus datos personales, y
le informa que tiene derecho a ejercitar los derechos ARCO (derecho de
acceso, rectificación, cancelación y oposición) de sus datos personales
mediante solicitud escrita a ***@***.***, adjuntando fotocopia del
D.N.I.

"En la vida hay tres cosas que no dan marcha atrás: la palabra pronunciada,
la flecha lanzada y la oportunidad perdida, ¿vamos a dejarla escapar?"

cparrapa, 2023-01-23:
I think the teacher meant something like this 
[image]

logix5, 2023-01-23:
These blocks are used to make a I2c general communication.

Scan devices is a program to check the device connected to I2C, we can add
this block too,  and PCA is hardware block to have more I2C ports

Do you want to add these type of blocks?

BR

-- 
Óscar Ferruz Romero
*Socio fundador.*
*Tel. 620.402.168*

 C/ Artajona, 17.  28039 - Madrid.
www.logix5.com

www.twitter.com/Logix5_SS <http://www.twitter.com/logix5_SS>

www.facebook.com/logix5

www.instagram.com/logix5_smart_solutions

Conforme dispone el Reglamento (UE) 2016/679 del Parlamento Europeo y del
Consejo, de 27 de abril de 2016, relativo a la protección de las personas
físicas en lo referente al tratamiento de datos personales y a la libre
circulación de estos, le informamos que los datos personales facilitados y
recogidos en este e-mail y sus documentos adjuntos serán tratados e
incluidos automatizadamente en los ficheros de datos personales bajo
responsabilidad de LOGIX5 SMART SOLUTIONS, S.L. B-86129699, donde se
recogen y almacenan los datos personales con la finalidad exclusiva de
formar parte en los procesos de selección de personal, bolsa de trabajo,
contratación o gestiones relacionadas con temas laborales que se lleven a
cabo. El interesado otorga su consentimiento para el tratamiento de sus
datos personales con la finalidad anteriormente mencionada. En el supuesto
de producirse alguna modificación de sus datos personales, le solicitamos,
nos lo comunique por escrito con la única finalidad de mantenerlos
actualizados.   LOGIX5 SMART SOLUTIONS, S.L. B-86129699 se compromete a
respetar su confidencialidad en el tratamiento de sus datos personales, y
le informa que tiene derecho a ejercitar los derechos ARCO (derecho de
acceso, rectificación, cancelación y oposición) de sus datos personales
mediante solicitud escrita a ***@***.***, adjuntando fotocopia del
D.N.I.

"En la vida hay tres cosas que no dan marcha atrás: la palabra pronunciada,
la flecha lanzada y la oportunidad perdida, ¿vamos a dejarla escapar?"

cparrapa, 2023-01-25:
someone requested in the community

logix5, 2023-01-25:
Hi Camilo

Can you ask this person what exactly he needs? I told you this because in
these blocks are using external hardware to have more pins using a I2C
multiplexer

We can try to add :-)

Let me know

-- 
Óscar Ferruz Romero
*Socio fundador.*
*Tel. 620.402.168*

 C/ Artajona, 17.  28039 - Madrid.
www.logix5.com

www.twitter.com/Logix5_SS <http://www.twitter.com/logix5_SS>

www.facebook.com/logix5

www.instagram.com/logix5_smart_solutions

Conforme dispone el Reglamento (UE) 2016/679 del Parlamento Europeo y del
Consejo, de 27 de abril de 2016, relativo a la protección de las personas
físicas en lo referente al tratamiento de datos personales y a la libre
circulación de estos, le informamos que los datos personales facilitados y
recogidos en este e-mail y sus documentos adjuntos serán tratados e
incluidos automatizadamente en los ficheros de datos personales bajo
responsabilidad de LOGIX5 SMART SOLUTIONS, S.L. B-86129699, donde se
recogen y almacenan los datos personales con la finalidad exclusiva de
formar parte en los procesos de selección de personal, bolsa de trabajo,
contratación o gestiones relacionadas con temas laborales que se lleven a
cabo. El interesado otorga su consentimiento para el tratamiento de sus
datos personales con la finalidad anteriormente mencionada. En el supuesto
de producirse alguna modificación de sus datos personales, le solicitamos,
nos lo comunique por escrito con la única finalidad de mantenerlos
actualizados.   LOGIX5 SMART SOLUTIONS, S.L. B-86129699 se compromete a
respetar su confidencialidad en el tratamiento de sus datos personales, y
le informa que tiene derecho a ejercitar los derechos ARCO (derecho de
acceso, rectificación, cancelación y oposición) de sus datos personales
mediante solicitud escrita a ***@***.***, adjuntando fotocopia del
D.N.I.

"En la vida hay tres cosas que no dan marcha atrás: la palabra pronunciada,
la flecha lanzada y la oportunidad perdida, ¿vamos a dejarla escapar?"

NeftaliVenancio, 2024-04-12:
@cparrapa  para poder leer más de un elemento i2c (Específicamente del caso del sensor TOF) es necesario usar un multiplexor de i2c, pues este sensor no tiene opciones para moficar la dirección de lectura, he podido resolver algunos bugs principalmente del código de Arduino, pero desconozco de que forma podría crear el bloque nuevo para realizar esta opción, pero con gusto podría colaborar en la parte del código en Arduino pero en JS me cuesta trabajo.