OttoDIY/blockly #147: Block color sensor problem

cparrapa, 2022-01-24:
some of the blocks don't seem to work https://vimeo.com/574938953/2400c22042
https://vimeo.com/574938953/2400c22042?fbclid=IwAR2rqtRa9bE8mnkHMCHwxueQ6hmoJxyUmfLmtHX-hgx1Zb-vyKACuvXpDts

logix5, 2022-01-25:
Hi I have reviewed and the blocks are working. This picture is the online
editor but is the same that the offline program:

[image: image.png]

BR

-- 
Óscar Ferruz Romero
*Socio fundador.*
*Tel. 620.402.168*

 C/ Artajona, 17.  28039 - Madrid.
www.logix5.com

www.twitter.com/Logix5_SS <http://www.twitter.com/logix5_SS>

www.facebook.com/logix5

www.instagram.com/logix5_smart_solutions

Conforme dispone el Reglamento (UE) 2016/679 del Parlamento Europeo y del
Consejo, de 27 de abril de 2016, relativo a la protección de las personas
físicas en lo referente al tratamiento de datos personales y a la libre
circulación de estos, le informamos que los datos personales facilitados y
recogidos en este e-mail y sus documentos adjuntos serán tratados e
incluidos automatizadamente en los ficheros de datos personales bajo
responsabilidad de LOGIX5 SMART SOLUTIONS, S.L. B-86129699, donde se
recogen y almacenan los datos personales con la finalidad exclusiva de
formar parte en los procesos de selección de personal, bolsa de trabajo,
contratación o gestiones relacionadas con temas laborales que se lleven a
cabo. El interesado otorga su consentimiento para el tratamiento de sus
datos personales con la finalidad anteriormente mencionada. En el supuesto
de producirse alguna modificación de sus datos personales, le solicitamos,
nos lo comunique por escrito con la única finalidad de mantenerlos
actualizados.   LOGIX5 SMART SOLUTIONS, S.L. B-86129699 se compromete a
respetar su confidencialidad en el tratamiento de sus datos personales, y
le informa que tiene derecho a ejercitar los derechos ARCO (derecho de
acceso, rectificación, cancelación y oposición) de sus datos personales
mediante solicitud escrita a ***@***.***, adjuntando fotocopia del
D.N.I.

"En la vida hay tres cosas que no dan marcha atrás: la palabra pronunciada,
la flecha lanzada y la oportunidad perdida, ¿vamos a dejarla escapar?"

cparrapa, 2023-01-23:
Hi @logix5 
The picture is not visible can you attach the .bloc example please?

logix5, 2023-01-23:
Hi @Camilo,

I don´t understand the problem, because I have seen the block codes and it
is showing in arduino mode.  Please let me know

-- 
Óscar Ferruz Romero
*Socio fundador.*
*Tel. 620.402.168*

 C/ Artajona, 17.  28039 - Madrid.
www.logix5.com

www.twitter.com/Logix5_SS <http://www.twitter.com/logix5_SS>

www.facebook.com/logix5

www.instagram.com/logix5_smart_solutions

Conforme dispone el Reglamento (UE) 2016/679 del Parlamento Europeo y del
Consejo, de 27 de abril de 2016, relativo a la protección de las personas
físicas en lo referente al tratamiento de datos personales y a la libre
circulación de estos, le informamos que los datos personales facilitados y
recogidos en este e-mail y sus documentos adjuntos serán tratados e
incluidos automatizadamente en los ficheros de datos personales bajo
responsabilidad de LOGIX5 SMART SOLUTIONS, S.L. B-86129699, donde se
recogen y almacenan los datos personales con la finalidad exclusiva de
formar parte en los procesos de selección de personal, bolsa de trabajo,
contratación o gestiones relacionadas con temas laborales que se lleven a
cabo. El interesado otorga su consentimiento para el tratamiento de sus
datos personales con la finalidad anteriormente mencionada. En el supuesto
de producirse alguna modificación de sus datos personales, le solicitamos,
nos lo comunique por escrito con la única finalidad de mantenerlos
actualizados.   LOGIX5 SMART SOLUTIONS, S.L. B-86129699 se compromete a
respetar su confidencialidad en el tratamiento de sus datos personales, y
le informa que tiene derecho a ejercitar los derechos ARCO (derecho de
acceso, rectificación, cancelación y oposición) de sus datos personales
mediante solicitud escrita a ***@***.***, adjuntando fotocopia del
D.N.I.

"En la vida hay tres cosas que no dan marcha atrás: la palabra pronunciada,
la flecha lanzada y la oportunidad perdida, ¿vamos a dejarla escapar?"

logix5, 2023-01-23:
Hi sorry
you can't

 see my image. I have attached. I hope this time works.

BR

-- 
Óscar Ferruz Romero
*Socio fundador.*
*Tel. 620.402.168*

 C/ Artajona, 17.  28039 - Madrid.
www.logix5.com

www.twitter.com/Logix5_SS <http://www.twitter.com/logix5_SS>

www.facebook.com/logix5

www.instagram.com/logix5_smart_solutions

Conforme dispone el Reglamento (UE) 2016/679 del Parlamento Europeo y del
Consejo, de 27 de abril de 2016, relativo a la protección de las personas
físicas en lo referente al tratamiento de datos personales y a la libre
circulación de estos, le informamos que los datos personales facilitados y
recogidos en este e-mail y sus documentos adjuntos serán tratados e
incluidos automatizadamente en los ficheros de datos personales bajo
responsabilidad de LOGIX5 SMART SOLUTIONS, S.L. B-86129699, donde se
recogen y almacenan los datos personales con la finalidad exclusiva de
formar parte en los procesos de selección de personal, bolsa de trabajo,
contratación o gestiones relacionadas con temas laborales que se lleven a
cabo. El interesado otorga su consentimiento para el tratamiento de sus
datos personales con la finalidad anteriormente mencionada. En el supuesto
de producirse alguna modificación de sus datos personales, le solicitamos,
nos lo comunique por escrito con la única finalidad de mantenerlos
actualizados.   LOGIX5 SMART SOLUTIONS, S.L. B-86129699 se compromete a
respetar su confidencialidad en el tratamiento de sus datos personales, y
le informa que tiene derecho a ejercitar los derechos ARCO (derecho de
acceso, rectificación, cancelación y oposición) de sus datos personales
mediante solicitud escrita a ***@***.***, adjuntando fotocopia del
D.N.I.

"En la vida hay tres cosas que no dan marcha atrás: la palabra pronunciada,
la flecha lanzada y la oportunidad perdida, ¿vamos a dejarla escapar?"

closed 2024-06-30 by cparrapa